realsenseai/librealsense #8522: RuntimeError: No device connected,but camera can be detected

amos0716, 2021-03-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | {(01.05.03.00)} | 
| Operating System & Version | {Linux (Ubuntu 18.04.5 LTS)}  | 
| Kernel Version (Linux Only)    | {(4.9.140-tegra) }                                        | 
| Platform                                 | {NVIDIA Jetson XAVIER NX}  |
| SDK Version                            | { 2.42.0 }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
I have build librealsense 2.42.0 from source using the instructions given in the https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md and L515 can work in  RealSense-Viewer
but when I use the examole code from /home/jetsonnx/librealsense/wrappers/python/example,It give the following error message like this:

```
Traceback (most recent call last):
  File "align-depth2color.py", line 25, in <module>
    profile = pipeline.start(config)
RuntimeError: No device connected
```

And I run 

> rs-enumerate-devices

The terminal shows this:
```
Device info: 
    Name                          : 	Intel RealSense L515
    Serial Number                 : 	f0232770
    Firmware Version              : 	01.05.03.00
    Recommended Firmware Version  : 	01.05.04.01
    Physical Port                 : 	/sys/devices/3610000.xhci/usb2/2-3/2-3.1/2-3.1:1.0/video4linux/video0
    Debug Op Code                 : 	15
    Product Id                    : 	0B64
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	L500
    Asic Serial Number            : 	0003aa2ea3e6
    Firmware Update Id            : 	0003aa2ea3e6

Stream Profiles supported by L500 Depth Sensor
 Supported modes:
    stream       resolution      fps       format   
    Confidence	  1024x768	@ 30Hz	   RAW8
...(Omit)
```
So I think camera(L515) can be detected,but i don't understand why I can't use it in example code or other .py code.
Could someone please help me diagnose this issue?

RealSenseSupport, 2021-03-12:
Hi amos0716,
did you build the python wrapper  on Jetson from scratch and use  it?
had you used old version python wrapper before?
if yes, please ensure the python use the new one not the old one which may not support L515 yet

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python

closed 2021-03-26 by RealSenseSupport

amos0716, 2021-04-22:
@RealSenseSupport thank you for your reply!
But I downloaded the newest python wrapper from github, it still show error message:

```
Traceback (most recent call last):
  File "align-depth2color.py", line 25, in <module>
    profile = pipeline.start(config)
RuntimeError: No device connected
```

Should I try to reinstall librealsense because I use L515 on Jetson XAVIER NX, and according to RealSense official github, RealSense  is not supported on Jetson XAVIER NX.

I installed librealsense by this page:
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

Thank you for your answer.