realsenseai/librealsense #8526: Questions about FORCE_RSUSB_BACKEND and HWM_OVER_XU, Connection issues.

rafaelspring, 2021-03-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 Series and L515 | 
| Operating System & Version |  Win10, Android  | 
| SDK Version                            |  All | 
| Language                            |  C++ | 
| Segment			|  Phones, Tablets | 

### Issue Description
We have a question about FORCE_RSUSB_BACKEND and HWM_OVER_XU config. options but first some background:

We sell a software product that supports RealSense sensors as input 3D cameras. We support Android and Windows. Some of our customers are seeing nasty connection issues, especially on Android and especially on non-Snapdragon devices. Our customers are (understandably) upset about this. The reported problems are manifold, but can be grouped:

1. Android doesn't see / recognize the RealSense camera as a USB device when connected via USB.

2. Android recognizes the camera and the RS Camera app works, but even though the device has a USB 3.1 compatible port only USB 2.0 modes are available.

3. Android recognizes the camera and the RS Camera app works with USB 3.1 modes, however when using our software librealsense fails to apply a critical camera / stream configuration and throws an error during the camera configuration phase of our software.

4. RS Camera app works and configuration of the camera in our SW works, however after asking libRS to deliver frames in the chosen configuration, our SW waits forever as libRS never delivers any frames.

5. Everything works as expected -- but only for about 5 - 20 seconds and at that point the camera is either reported as disconnected or one or both of the color / depth streams stop delivering data and as a consequence our software appears to be "frozen".

Sometimes a device shows one of the symptoms with a D400-based camera but works fine with the L515, sometimes it's the other way round. Also, we already know the "trick" of flipping the (supposed-to-be-symmetrical) USB C cable around and sometimes that fixes things.

### Question:
Do the FORCE_RSUSB_BACKEND and HWM_OVER_XU have any influence on connection stability on Android and Windows? If yes, how? In general: Which combination of options do you recommend for maximizing connection stability / minimizing above-mentioned issues?

### Additional Questions:
Quoting the relevant sections from from https://github.com/IntelRealSense/librealsense/wiki/Build-Configuration:
| | |
|---------------------------------|------------------------------------------- |
| FORCE_LIBUVC | Setting this flag to true will configure the underlying backend to use libuvc. This backend is default for Mac OS, but can be configured on other OS-s as well. libuvc can serve as a more robust alternative to the native backend, however, it has well known limitations and is not recommended to be used in end-products (please use FORCE_RSUSB_BACKEND cmake flag instead) | 
| FORCE_WINUSB_UVC | Setting this will configure librealsense to use Windows backend based on WinUSB rewrite of the UVC protocol, instead of Media Foundation. Requires installation of a special driver provided inside the Win7 SDK installer (use FORCE_RSUSB_BACKEND instead) |
| HWM_OVER_XU | When this option is enabled, hardware commands will be dispatched through extension-unit (XU) transfer protocol. When disabled (and assuming the hardware supports it) the SDK will talk to the hardware directly using libusb / WinUSB |
| ANDROID_USB_HOST_UVC | Build UVC backend for Android (use FORCE_RSUSB_BACKEND instead) |
| FORCE_RSUSB_BACKEND | Equivalent to FORCE_LIBUVC, ANDROID_USB_HOST_UVC and FORCE_WINUSB_UVC. Renamed for consistency across platforms |

**FORCE_LIBUVC:**
> libuvc can serve as a more robust alternative to the native backend, however, it has well known limitations and is not recommended to be used in end-products

**What are the limitations? What does "more robust" mean?**

**FORCE_WINUSB_UVC:**
> Requires installation of a special driver provided inside the Win7 SDK installer

**So is this even an option for Windows 10?**

**HWM_OVER_XU: What are the implications of enabling it? Does it have any influence on connection stability?**

RealSenseSupport, 2021-03-18:
@rafaelspring LibUVC and WinUSB are deprecated from the build in SDK 2.30+, replaced by RSUSB. For more details please refer to https://dev.intelrealsense.com/docs/build-configuration 

According to https://github.com/IntelRealSense/librealsense/issues/4132#issuecomment-498000053 , HWM_OVER_XU is somewhat "internal" and mostly happens internally when debugging new platforms / devices.

RealSenseSupport, 2021-03-31:
@rafaelspring Any other questions from your side? Looking forward to your update. Thanks!

RealSenseSupport, 2021-04-06:
@rafaelspring Do you have any other questions about this? Please note that the ticket will be closed if we don't hear from you for another 7 days. Thanks!

RealSenseSupport, 2021-04-20:
@rafaelspring Sorry that we don't hear from you for weeks. Will close the issue at this point. Please feel free to create another one if you still have questions or issues. Thanks!

closed 2021-04-20 by RealSenseSupport