realsenseai/librealsense #8528: Retrieving Depth Frame from recording

Zumbalamambo, 2021-03-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   |05.12.11.00| 
| Operating System & Version |  Ubuntu 20.0.4 | 
| Platform                                 | PC  |
| Language                            |  python        | 
| Segment			|  others            | 

I have used the following code to record,
```

import pyrealsense2 as rs
import numpy as np
import cv2

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

color_path = 'V00P00A00C00_rgb.avi'
depth_path = 'V00P00A00C00_depth.avi'
colorwriter = cv2.VideoWriter(color_path, cv2.VideoWriter_fourcc(*'XVID'), 30, (640, 480), 1)
depthwriter = cv2.VideoWriter(depth_path, cv2.VideoWriter_fourcc(*'XVID'), 30, (640, 480), 1)

pipeline.start(config)

try:
    while True:
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        if not depth_frame or not color_frame:
            continue

        # convert images to numpy arrays
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)

        colorwriter.write(color_image)
        depthwriter.write(depth_colormap)

        cv2.imshow('Stream', depth_colormap)

        if cv2.waitKey(1) == ord("q"):
            break
finally:
    colorwriter.release()
    depthwriter.release()
    pipeline.stop()

```

But, how do I read the depth data from the depth video V00P00A00C00_depth?

MartyG-RealSense, 2021-03-08:
Hi @Zumbalamambo  As you are using Python, cv and .avi, I tried to find a video conversion method that fit these criteria.  I located a discussion that offers a range of different techniques for accessing video frames with Python code.

The link below directs to a section of the discussion that suggests a method for .avi format file reading, though the whole of the lengthy discussion is worth browsing from the beginning.

https://stackoverflow.com/a/52463543

*********************

I also recall that it should be possible to use the SDK's **software_device** interface indirectly to access video-file frames.

https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device

A RealSense team member suggested in Point 3 of the comment linked to below that .avi frames could be read by OpenCV and then fed to software_device.

https://github.com/IntelRealSense/librealsense/issues/2249#issuecomment-413492867

Zumbalamambo, 2021-03-08:
@MartyG-RealSense 

Thank you! Ofcourse I have read it with  the following code,

```
import cv2

cap = cv2.VideoCapture("V00P00A00C00_depth.avi")
while (1 == 1):
    ret, colormap_frame = cap.read()
    if ret:
        cv2.imshow("res", colormap_frame)
    if  cv2.waitKey(1) == ord('q'):
        break
cap.release()
cv2.destroyAllWindows()
```

But what I'm interested is how can I retrieve the depth data from the `colormap_frame` ?

MartyG-RealSense, 2021-03-08:
There is a suggested process for converting a cv::mat to the SDK's rs2::frame, though I'm not sure if your **colormap_frame** is a cv::mat.

https://github.com/IntelRealSense/librealsense/issues/2634#issuecomment-433904079

Below is an alternative approach for converting cv:mat to rs2::frame.

https://stackoverflow.com/questions/52585921/convert-opencv-mat-to-frame-type-of-realsense-in-librealsense-sdk 

If you require further suggestions, please do let me know.

MartyG-RealSense, 2021-03-08:
Do you mean converting numpy to a RealSense depth frame please?

Zumbalamambo, 2021-03-08:
Here is my problem,

The recorded bag file is of 158GB in terms of file size. It is practically impossible to transfer it or download it during this period of lockdown.
So, I have written the rgb frames as an avi file and the depth images (encoded as a color map) as another avi file.

I would like to know how I can restrore the depth data after reading the frames from the colormap. I have successfully read the videofile that has the colormap in which the depth data is available. 

In [this](https://github.com/IntelRealSense/librealsense/issues/8528#issuecomment-793000402) code , I'm the `color_frame` is the colormap that I have initially encoded. It is in the numpy format. How do I retrieve depth data from it by converting it into the realsense depth format?

MartyG-RealSense, 2021-03-08:
If bag file size is the primary reason for the avi based system, then a solution may be to design your project around **ROS** and its **rosbag record** system instead,sand use it with the RealSrnse ROS wrapper

https://github.com/IntelRealSense/realsense-ros

http://wiki.ros.org/rosbag/Commandline#rosbag_record

ROS has a bag record function called **split** that enables the recording to be automatically split into multiple smaller files when a certain defined condition such as s length of time or a file size is reached, closing the current recording and starting a new one that carries on from that point in the recording session.

Zumbalamambo, 2021-03-09:
@MartyG-RealSense 

I recorded using both Realsense and ZED.  ZED also saves RGB, DEPTH, POSE information but its file size was only 7 GB while intel's bag file is almost 120 GB!  It's not suitable to debug with ease or to handle fragments of files by breaking the complete pieces into multiple smaller pieces of bags.

closed 2021-03-09 by Zumbalamambo

MartyG-RealSense, 2021-03-10:
The SDK's **Sample Data** page states that SDK bag files are large because they contain uncompressed and unfiltered camera data.

https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md

If you are recording your bag in RealSense Viewer @Zumbalamambo then do you have a noticably smaller file size if you enable 'Always Compress' bag compression on the **Playback & Record** section of the Viewer's **Settings** interface, please?

The Settings window can be accessed by going to the gear-wheel icon in the top corner of the Viewer and selecting the **Settings** drop-down menu option.

[image]

[image]

tortoisehare, 2022-02-24:
@Zumbalamambo I know this was awhile ago, but did you figure out how to convert the numpy colormap_frame into the realsense depth format?  Or did you find a different way to retrieve the depth information?  I'm running into a similar issue with the large .bag sizes and I'd love to use these .avi files, but I need to make sure I can access the depth information.

Unfortunately, compressing the files isn't a good option since the program says that may lead to frame drops.  And splitting into smaller .bag sizes wouldn't resolve the issue of storing and uploading a large amount of total data.