realsenseai/librealsense #8529: How to confirm D4xx cameras are in Slave mode?

bobluoluo, 2021-03-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  Ubuntu 18.04 - 4.18  or Raspbian img(https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras)                                         | 
| Platform                                 | PC/Raspberry Pi  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            | c/c++                          | 

### Issue Description
So, I set the external sync mode to 2(slave mode). How do I verify sync mode is set? 
[image]

According to the document, "When the Genlock mode is selected, the camera will wait indefinitely for a trigger." The problem is, the camera DOES NOT wait indefinitely in slave mode when sync pulse is kicked in; it runs as usual. So, how can I verify if sync is doing its job? 

[image]

  NO SYNC in this picture, but the camera still runs.
[image]

MartyG-RealSense, 2021-03-08:
Hi @bobluoluo   Mode '2' (Slave) on the Inter Cam Sync Mode settings is not genlock.  Mode '3' (Full Slave) and above belongs to the External Synchronization (genlock) system.

Modes 1 (Master) and 2 (Slave) originate from an earlier hardware sync system that does not have Mode 3 and above.  Mode 2 (Slave) behaves differently from the genlock system.  In Mode 2, the camera listens on each frame for a sync trigger pulse from a Master camera or from a signal generator, and if it does not detect that pulse within a certain time period then it stops listening and initiates capture independently (unsynced).  Mode 2 does not wait indefinitely.

The hardware sync system that uses Mode 1 and 2 is described in the Intel white-paper document linked to below.  This system is mature and validated by Intel.   The more recent genlock system (Mode 3+) is not considered mature, but you are free to experiment with it.

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration

In regard to verifying whether hardware sync is occurring or whether the streams are unsynched, Section 3 of the paper explains that you can observe whether timestamps are drifting apart over time to confirm sync.  If there is drift then hardware sync is active.  If the timestamps do not drift then the system is unsynced.  This paradox is explained in Section 3 in the paragraph with the bolded text **Now to the somewhat counter intuitive aspect of time stamps**.

Sync can also be validated by using high-speed LED panels, though it is expected that most users will not need to use this high-tech apprpoach.

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration#section-g-hw-sync-validation

https://support.intelrealsense.com/hc/en-us/community/posts/360049401673/comments/360012819533

sam598, 2021-03-08:
@MartyG-RealSense there still seems to be a lot of confusion about Mode '3' (Full Slave) and the Genlock Modes '4+'

According to the whitepaper and @RealSenseSupport in this issue https://github.com/IntelRealSense/librealsense/issues/8419 Mode '3' is intended to operate exactly the same as Mode '2', except that it is supposed to also sync the color camera as well, hence "Full".

Unfortunately Mode '3' is just a placeholder for now as that feature has not been implemented.

Modes '4+' are the "genlock" modes that allow arbitrary framerates from an external source. 

Because you have recently been describing Modes '4+' as "experimental" and Mode '3' is "unfinished" I sort of get lumping them together. But because Modes '4+' have additional limitations I wanted to make sure that I am understanding all of this correctly.

Thanks.

MartyG-RealSense, 2021-03-08:
@sam598   I have no further information to add beyond the already stated position that Mode 1 and 2 belong to the mature and validated hardware sync system.  You are free to experiment with Mode 3+ but there is no technical support that I can provide for that.

MartyG-RealSense, 2021-03-14:
Hi @bobluoluo  Do you require further assistance with this case, please?  Thanks!

bobluoluo, 2021-03-15:
Thanks you all for the helpful information about the sync. Frames under sync mode 4 can be triggered individually by a 30fps sync signal. However it realsense-viewer shows the frame rate is @15fps. What can I do to fix this.
[image]

MartyG-RealSense, 2021-03-15:
Hi @bobluoluo   If you are using a master and slave camera under the genlock system (and I see in the image above that you are using Mode 4) then the slave camera should use a higher FPS than the master camera.  This is described in the section of the genlock paper linked to below.

https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras#section-2-principles-of-operation

To quote this section:

"In order to have all cameras stream at 30fps, for example, the Master must be set to 30fps, and all Genlock cameras must be natively set to 90fps modes with burst count set to 1.  The master camera will then deliver the trigger at 30fps, and all Genlock cameras will trigger in sync with the master". 

[image]

bobluoluo, 2021-03-15:
I set the trigger to 60 fps, native 60 fps . And it gives me this. 
[image]

I guess my question is how can I get 30 fps under mode 4?

bobluoluo, 2021-03-15:
I can't believe I missed such important text. I must be blind. :-) 
So, I tried native mode 60fps + trigger 30fps and 60fps + trigger 60fps. Neither of them worked. 
How do I setup a 30fps capture under mode 4 correctly?

bobluoluo, 2021-03-15:
> I can't believe I missed such important text. I must be blind. :-)
> So, I tried native mode 60fps + trigger 30fps and 60fps + trigger 60fps. Neither of them worked.
> How do I setup a 30fps capture under mode 4 correctly?

Little update. Previous comment, its experiment was made on a Raspberry PI with RTSP solution. 
The good news is the configuration of native 60 fps + pulse 30 fps is working under mode 4 with USB(direct connection) solution.

MartyG-RealSense, 2021-03-15:
If you are using the RTSP protocol that is part of Intel' open-source ethernet networking system that uses **rs-server** then rs-server has some steaming limitations at the time of writing this.  In order to use 60 FPS, a resolution of **480x270** should be used.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-3-6-software-limitations

A new networking extension called **LRS-Net 2.0** will be in an upcoming SDK version that has support for a greater range of resolutions, formats and FPs speeds.

https://github.com/IntelRealSense/librealsense/issues/8372

bobluoluo, 2021-03-15:
Do we have a release date? I can't wait. :-)

MartyG-RealSense, 2021-03-15:
I do not have a release date, though its developer welcomes people testing it before its release.  Their LRS-Net 2.0 fork of librealsense is in the link below.

https://github.com/IntelRealSenseArchive/lrs-net/tree/apuzhevi_lrsnet

bobluoluo, 2021-03-15:
Just encountered a couple of compiler error. Do I need some extra dependencies? What flags should I use (btw, master  compiles just fine.)

[image]

My cmake cmoonad is : cmake .. -DBUILD_NETWORK_DEVICE=ON -DFORCE_RSUSB_BACKEND=ON

MartyG-RealSense, 2021-03-15:
What method did you use to compile the source code please?  

Did you download the source as a zip file using the **Code** button on its GitHub, extract the folder and then **mkdir build && cd build**?

bobluoluo, 2021-03-15:
yes, I cloned the fork branch and it gives me compiler error. 

The "master-branch" under the primary branch https://github.com/IntelRealSense/librealsense has no build issue is what I meant. Sorry about the confusion.

MartyG-RealSense, 2021-03-15:
I will tag the LRS-Net developer @apuzhevi into this discussion to seek input on how to compile his LRS-equipped fork of librealsense.  Thanks very much @bobluoluo

bobluoluo, 2021-03-16:
> yes, I cloned the fork branch using the link you gave me and it outputs compiler error during build. e.g. make . -j4

OKay. I think I found the problem. I made a mistake. The clone command provided by github does not switch to the default branch automatically. It is still pointing at the very old master branch when it was forked. 
[image]

the code builds successfully on branch apuzhevi_lrsnet

bobluoluo, 2021-03-16:
I'm now running the new rs-server. How do I solve this issue? 
[image]

MartyG-RealSense, 2021-03-16:
Hi @bobluoluo   Excellent to hear that you built the fork successfully.

I guess that the issue you are referring to is the side-panel options showing values of zero?  If so, what do the drop-down menus look like when opened (if they are openable)?

bobluoluo, 2021-03-16:
no items in the drop downs
[image]
[image]
[image]

MartyG-RealSense, 2021-03-16:
Did you set the static IP in the Pi's secure terminal, which is a step in the ethernet white-paper that comes before setting the camera's Network Device number?

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-2-6-setting-static-ip

bobluoluo, 2021-03-16:
Sorry again, I forgot to update the code on the client side too. Now I got this.

Now I got this. There might be a bug in the UI I see two of the same IP camera interface in the device list. 

[image]

MartyG-RealSense, 2021-03-16:
Thanks very much for the illustrated reports of your progress.  If you start the streams of one of the listed cameras, does it stream correctly please?

bobluoluo, 2021-03-16:
I tried starting a depth sensor with following settings and it failed.

[image]
[image]

bobluoluo, 2021-03-16:
FYI, at the server side(the machine running rs-server), it does show it support this profile. 
[image]

And couple of error/warning at server side.
[image]

MartyG-RealSense, 2021-03-16:
If you have two D455 listed, does streaming work if you close the lower one in the side-panel?

[image]

bobluoluo, 2021-03-16:
yes, the x button works.

MartyG-RealSense, 2021-03-17:
Hi @bobluoluo  Do you have new progress to report since your problem with enabling a sensor, please?

bobluoluo, 2021-03-17:
I can't turn on the camera on any allowed profile. So I'm writing up some code for testing. 

in the meantime, I got some error when running  sudo make install  

CMake Error at src/network/cmake_install.cmake:128 (file):
  file INSTALL cannot find
  "/home/bob/dev/lrs-net/build/src/ethernet/realsense2-netConfig.cmake".
Call Stack (most recent call first):
  cmake_install.cmake:132 (include)

MartyG-RealSense, 2021-03-17:
I will seek advice from the LRS-Net developer about your current problems.  Thanks very much for your patience.

MartyG-RealSense, 2021-03-18:
Hi @bobluoluo   I just wanted to provide a quick update that the developer of RS2-Net is looking at the information that you have kindly posted above.

bobluoluo, 2021-03-22:
btw, the "make install" script seems to be broken too. 

-- Installing: /usr/local/include/librealsense2-net/rs_net.hpp
-- Installing: /usr/local/lib/cmake/realsense2-net/realsense2-netTargets.cmake
-- Installing: /usr/local/lib/cmake/realsense2-net/realsense2-netTargets-noconfig.cmake
CMake Error at src/network/cmake_install.cmake:128 (file):
  file INSTALL cannot find
  "/home/bob/dev/lrs-net/build/src/ethernet/realsense2-netConfig.cmake".
Call Stack (most recent call first):
  cmake_install.cmake:132 (include)

Makefile:85: recipe for target 'install' failed
make: *** [install] Error 1

MartyG-RealSense, 2021-03-22:
Thanks very much @bobluoluo for the update of your feedback!

bobluoluo, 2021-03-23:
I actually have two questions related to the new rs-server. 1. What is the approximate release date for the new rs-server/receiver, a week or two? 2. Will it provide some synchronization method over the network with like NTP or PTP or something else. The reason why I ask is I'm evaluating if I can build a reliable Realsense project based on it. Thanks.

MartyG-RealSense, 2021-03-23:
1.  There is not currently a release date for LRS-Net 2.0.  Given that SDK 2.43.0 has just been released, it would not be available in an SDK release at least during the next month from the time of writing this.

2.  There is not a mention about synchronization on the advance-released feature list.

https://github.com/IntelRealSense/librealsense/pull/8343

bobluoluo, 2021-04-01:
yes, please help me. How do I fix this.  it was compiled with make -j1
[image]

MartyG-RealSense, 2021-04-01:
@bobluoluo  I will contact the LRS-Net developer again.

bobluoluo, 2021-04-05:
I downloaded the pre-release 2.44.0. Compiler error is fixed but the rs-server is no good.
[image]

bobluoluo, 2021-04-05:
usb has no issue. firmware is the latest.
[image]

bobluoluo, 2021-04-05:
[image]

MartyG-RealSense, 2021-04-06:
I note that the D455 camera above is listed in the options side panel as D455.  Have you tried it configured as a Network Camera like in your previous tests, please?

[image]

bobluoluo, 2021-04-06:
I configured a D455 on a Raspberry PI and ran rs-server(V2.44.0). I launched realsense-viewer(v2.44.0) on a server. Like this.
[image]
[image]
[image]
[image]

This is the error I got on the server
[image]

bobluoluo, 2021-04-06:
In my previous previous test on lrs-net 2.0.  Server sees two cameras even though I attached  only one D455 on the Raspberry PI.
[image]

And then,  when I enable a camera either stereo or rgb camera, the rs-server crashes with no useful debug info but segfault.

bobluoluo, 2021-04-06:
In summary, I tested both lrs-net 2.0 and v2.44.0 on the same hardware. Unfortunately, none of them works without issue.
I also tested lrs-net-dev branch. It has a compiler error.

MartyG-RealSense, 2021-04-07:
Another RealSense user encountered the same problem with accessing the network camera in the RealSense Viewer today, in SDK 2.43.0 in that particular case.

https://github.com/IntelRealSense/librealsense/issues/8766

MartyG-RealSense, 2021-04-08:
Hi @bobluoluo   Regarding your errors with rs-server in https://github.com/IntelRealSense/librealsense/issues/8529#issuecomment-814275813 - the original version of rs-server in the SDK supports a limited range of resolutions listed in the link below.  

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-3-6-software-limitations

424x240 supports 6, 15, 30 and 60 FPS for the **RGB** stream.  For Z16 depth, **480x270** should be used with FPS speeds 6, 15, 30 or 60.

MartyG-RealSense, 2021-04-08:
Also, please do not use genlock with rs-server and instead set Inter Cam Sync Mode to '0' in the Viewer if you are using a mode other than 0.

bobluoluo, 2021-04-08:
@MartyG-RealSense  The original limited version has no issue. Please don't get confused. I'm able to run it with or without gen-lock mode enabled on the original code. The problem is in the lrs-net and v2.44.0-pre. Either mode 0 or not, the program crashes in lrs-net, v2.44.0. We are looking forward to using this new lrs-net with more resolution options.

bobluoluo, 2021-04-08:
Please let know if there is a plan of fixing the bugs in lrs-net or v2.44.0-pre. Thanks

mirkocomparetti-synesis, 2021-04-09:
We have the same problem with the 2.44 viewer; 2.42 works fine. It got broken from 2.43

MartyG-RealSense, 2021-04-09:
Hi @bobluoluo and @mirkocomparetti-synesis   This case and the recent problems encountered with 2.43 / 2.44 has already been highlighted to the networking developer.   I cannot provide a time estimate for when a response will be received though.

bobluoluo, 2021-04-13:
The other thing I want to emphasis is I found out I can't launch more than one net_device in neither old code and new code.

MartyG-RealSense, 2021-04-19:
Hi @bobluoluo  Have you made further progress in this case yet?  Thanks!

bobluoluo, 2021-04-21:
nope

MartyG-RealSense, 2021-04-22:
Unfortunately I believe that there is nothing more that I can add to this case as it requires the input of the networking developer and I cannot compel them to respond.  I do apologize.

bobluoluo, 2021-04-26:
Thank you for the patient assistance.

MartyG-RealSense, 2021-05-02:
I thought I'd bring to your attention an item in the release notes of the new **2.45.0** SDK that was released today.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.45.0

*******************

[LRS-Net]  Fix connecting to network device with realsense-viewer

There was changes introduced in Feb 15 in model-views.cpp that breaks connections to network devices from realsense-viewer. 
 This PR fixes this. 

https://github.com/IntelRealSense/librealsense/pull/8751

mirkocomparetti-synesis, 2021-05-04:
Hello, at least this works.
Still compiling source with python does not generate the full library (i.e. missing pipeline object) on arm...

MartyG-RealSense, 2021-05-04:
Hi @mirkocomparetti-synesis   Could you provide more information about your Python problem please, including any error message that you receive?

mirkocomparetti-synesis, 2021-05-04:
Hello @MartyG-RealSense sure! https://github.com/IntelRealSense/realsense-ros/issues/1804#issuecomment-828210420 here it is!

[11 more replies]

closed 2021-07-28 by MartyG-RealSense