realsenseai/librealsense #8531: Build failed in windows 10 with option _BUILD_NETWORK_DEVICE

seliber, 2021-03-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10)   | 
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Follow the guide, try to build source in windows 10 machine, https://dev.intelrealsense.com/docs/compiling-librealsense-for-windows-guide

Env: visual studio 2019, 
Windows: windows 10, version 1909(OS build 18363.1377)

But it failed in realsense2-net project. I met lots of errors. 
e.g. cannot find "ifaddrs.h", cannot find "if.h", both the two header files are in linux environment, why does it needed in windows build?

If remove the file "GroupsockHelper.cpp" which referred the two header file, build the project again.  Will met lots of link error as below:

`1>AC3AudioStreamFramer.obj : error LNK2001: unresolved external symbol "int __cdecl gettimeofday(struct timeval *,int *)" (?gettimeofday@@YAHPEAUtimeval@@PEAH@Z)
1>ADTSAudioFileSource.obj : error LNK2001: unresolved external symbol "int __cdecl gettimeofday(struct timeval *,int *)" (?gettimeofday@@YAHPEAUtimeval@@PEAH@Z)
1>AMRAudioFileSource.obj : error LNK2001: unresolved external symbol "int __cdecl gettimeofday(struct timeval *,int *)" (?gettimeofday@@YAHPEAUtimeval@@PEAH@Z)
1>  Hint on symbols that are defined and could potentially match:
1>    "public: static void __cdecl el::base::utils::DateTime::gettimeofday(struct timeval *)" (?gettimeofday@DateTime@utils@base@el@@SAXPEAUtimeval@@@Z)
1>Groupsock.obj : error LNK2019: unresolved external symbol "int __cdecl readSocket(class UsageEnvironment &,int,unsigned char *,unsigned int,struct sockaddr_storage &)" (?readSocket@@YAHAEAVUsageEnvironment@@HPEAEIAEAUsockaddr_storage@@@Z) referenced in function "public: virtual bool __cdecl Groupsock::handleRead(unsigned char *,unsigned int,unsigned int &,struct sockaddr_storage &)" (?handleRead@Groupsock@@UEAA_NPEAEIAEAIAEAUsockaddr_storage@@@Z)
1>GenericMediaServer.obj : error LNK2001: unresolved external symbol "int __cdecl readSocket(class UsageEnvironment &,int,unsigned char *,unsigned int,struct sockaddr_storage &)" (?readSocket@@YAHAEAVUsageEnvironment@@HPEAEIAEAUsockaddr_storage@@@Z)
1>RTPInterface.obj : error LNK2001: unresolved external symbol "int __cdecl readSocket(class UsageEnvironment &,int,unsigned char *,unsigned int,struct sockaddr_storage &)" (?readSocket@@YAHAEAVUsageEnvironment@@HPEAEIAEAUsockaddr_storage@@@Z)
1>RTSPClient.obj : error LNK2001: unresolved external symbol "int __cdecl readSocket(class UsageEnvironment &,int,unsigned char *,unsigned int,struct sockaddr_storage &)" (?readSocket@@YAHAEAVUsageEnvironment@@HPEAEIAEAUsockaddr_storage@@@Z)
1>Groupsock.obj : error LNK2019: unresolved external symbol "bool __cdecl writeSocket(class UsageEnvironment &,int,struct sockaddr_storage const &,unsigned char,unsigned char *,unsigned int)" (?writeSocket@@YA_NAEAVUsageEnvironment@@HAEBUsockaddr_storage@@EPEAEI@Z) referenced in function "public: virtual bool __cdecl OutputSocket::write(struct sockaddr_storage const &,unsigned char,unsigned char *,unsigned int)" (?write@OutputSocket@@UEAA_NAEBUsockaddr_storage@@EPEAEI@Z)
1>Groupsock.obj : error LNK2019: unresolved external symbol "bool __cdecl writeSocket(class UsageEnvironment &,int,struct sockaddr_storage const &,unsigned char *,unsigned int)" (?writeSocket@@YA_NAEAVUsageEnvironment@@HAEBUsockaddr_storage@@PEAEI@Z) referenced in function "public: virtual bool __cdecl OutputSocket::write(struct sockaddr_storage const &,unsigned char,unsigned char *,unsigned int)" (?write@OutputSocket@@UEAA_NAEBUsockaddr_storage@@EPEAEI@Z)
`

seliber, 2021-03-10:
Thanks @MartyG-RealSense for your response, the story is:

1.  I built the project in raspberry pi with BULID_NETWORK_DEVICE option, and ran "rs-server" in raspberry pi successfully. 
2. Then I connected my Pi with PC, tried to get the stream in PC side. 
3. I built the source in PC side without BUILD_NETWORK_DEVICE, then when I tried to add a network device, it failed with notification, "you need to build with BUILD_NETWORK_DEVICE, blablabla"
4. So I built the source in PC with the option, then it failed again and again.

My aiming is trying to write a streaming server in raspberryPi and receiving the stream in PC side, so I thought I could integrate **realsense2-net** in my project.

seliber, 2021-03-10:
@MartyG-RealSense , BTW, another finding, After I change the live555 package from latest to version 2020.11.20, it can built successfully now. 

So, if libRealsense cannot be built with the latest live555, I would suggest to change the scripts to use a verified version of live555 library.

MartyG-RealSense, 2021-03-10:
Great to hear that you found a solution @seliber  - I will pass your feedback to the librealsense networking system's developer.

seliber, 2021-03-11:
@MartyG-RealSense the built version cannot worked well in my test, could you please provide the cmake commands to build the release version for windows?

The installed version didn't contain realsense2-net.dll..

MartyG-RealSense, 2021-03-11:
Does the advice provided in your other case https://github.com/IntelRealSense/librealsense/issues/8554#issuecomment-796674966 help with the question above, please?

seliber, 2021-03-12:
This issue can be closed, I think we can switch to another issue then. Thanks.

closed 2021-03-12 by seliber