realsenseai/librealsense #8532: Question about get RGB file from record

Jonapple, 2021-03-09:
| Camera Model                       | { D455 } | 
| Firmware Version                   | 05.12.10.00 | 
| Operating System & Version |   MacOS   | 
| Platform                                 | Mac XCode  |
| SDK Version                            |  2.2.42.0  | 
| Language                            |  C++ }                          | 
| Segment			|  vision  | 

I'm new to this and got some question about how to use it.

I converted the CMake into an Xcode Project and able to build it to get the GUI menu.

In the GUI menu, I got two streams in the 2D model, one is Depth and another one is Infrared
[image]

I couldn't find any option to change one of the streams into RGB.

I tried to record by using "Start_record" at the bottom, 
[image]
it saved:

png files for depth
raw files for infrared
raw files for depth
CSV files for infrared
CSV files for Depth

But there are no files for colour, how can I get that??

Thanks in advance.

MartyG-RealSense, 2021-03-09:
Hi @Jonapple   Your XCode project is based on the SDK program **Depth Quality Tool**, which does not have an RGB mode.

To have RGB options available, I would recommend doing an XCode conversion of the **RealSense Viewer** program instead.  As these two programs have very similar architecture, it should hopefully be a straightforward process.

[image]

MartyG-RealSense, 2021-03-16:
Hi @Jonapple  Do you require further assistance with this case, please?   Thanks!

Jonapple, 2021-03-18:
Problem solved, thanks

closed 2021-03-18 by Jonapple