realsenseai/librealsense #8546: ERROR [546912059776] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: d rs2_deproject_pixel_to_point(float*, const rs2_intrinsics*, const float*, float)

danielsmith185742, 2021-03-10:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 05.12.11.00 | 
| Operating System & Version |  Ubuntu 18.04.2 LTS  | 
| Kernel Version (Linux Only)    | 4.9.140-tegra                                        | 
| Platform                                 | NVIDIA Jetson TX2 |
| SDK Version                            | librealsense2 v2.42.0                          | 
| Language                            |  RealSense ROS v2.2.22                       | 
|ROS distro | Melodic  | 

SUMMARY
========

I am failing to get the realsense D435 working with RealSense ROS v2.2.22.  I am clearly doing something badly wrong and I am hoping somebody can point me in the right direction.

I followed the instructions here to build the librealsense2 SDK...

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

When I run,

roslaunch realsense2_camera rs_camera.launch align_depth:=true initial_reset:=true

I get the following output and errors...

Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

/opt/ros/melodic/share/realsense2_camera/launch/rs_camera.launch
started roslaunch server http://192.168.2.100:38191/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
[... 85 lines cut ...]
 10/03 18:33:43,323 WARNING [547465716096] (backend-v4l2.cpp:1057) Incomplete frame received: Incomplete video frame detected!
Size 65784 out of 814335 bytes (8%)
 10/03 18:33:43,337 WARNING [547465716096] (backend-v4l2.cpp:1057) Incomplete frame received: Incomplete video frame detected!
Size 214016 out of 814335 bytes (26%)
 10/03 18:33:43,350 ERROR [547474108800] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: d rs2_deproject_pixel_to_point(float*, const rs2_intrinsics*, const float*, float)
 10/03 18:33:43,352 ERROR [547474108800] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: d rs2_deproject_pixel_to_point(float*, const rs2_intrinsics*, const float*, float)
 10/03 18:33:43,353 ERROR [547474108800] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: d rs2_deproject_pixel_to_point(float*, const rs2_intrinsics*, const float*, float)
 10/03 18:33:43,354 WARNING [547474108800] (backend-v4l2.cpp:1057) Incomplete frame received: Incomplete video frame detected!
Size 65784 out of 614655 bytes (10%)
 10/03 18:33:43,381 ERROR [547474108800] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: d rs2_deproject_pixel_to_point(float*, const rs2_intrinsics*, const float*, float)
 10/03 18:33:43,474 ERROR [547474108800] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: d rs2_deproject_pixel_to_point(float*, const rs2_intrinsics*, const float*, float)
 10/03 18:33:43,506 WARNING [547474108800] (backend-v4l2.cpp:1057) Incomplete frame received: Incomplete video frame detected!
Size 181248 out of 614655 bytes (29%)
 10/03 18:33:43,538 WARNING [547465716096] (backend-v4l2.cpp:1057) Incomplete frame received: Incomplete video frame detected!
Size 65784 out of 814335 bytes (8%)
 10/03 18:33:43,547 WARNING [547465716096] (backend-v4l2.cpp:1057) Incomplete frame received: Incomplete video frame detected!
Size 377856 out of 814335 bytes (46%)
 10/03 18:33:43,551 WARNING [547474108800] (backend-v4l2.cpp:1057) Incomplete frame received: Incomplete video frame detected!
Size 312320 out of 614655 bytes (50%)
 10/03 18:33:43,555 WARNING [547465716096] (backend-v4l2.cpp:1057) Incomplete frame received: Incomplete video frame detected!
Size 65784 out of 814335 bytes (8%)
 10/03 18:33:43,586 WARNING [547465716096] (backend-v4l2.cpp:1057) Incomplete frame received: Incomplete video frame detected!
Size 541696 out of 814335 bytes (66%)
 10/03 18:33:43,606 WARNING [547465716096] (backend-v4l2.cpp:1057) Incomplete frame received: Incomplete video frame detected!
Size 377856 out of 814335 bytes (46%)

danielsmith185742, 2021-03-12:
Hi,

Thanks for your reply, I agree this seems to be related to the librealsense SDK and I wonder if anybody out there is running v2.42 on a Jetson TX2?

Regarding the cable I am using the one provided with the RealSense and it is in good condition. The camera and cable work fine if I plug them into my Windows 10 PC and use the RealSense Viewer. 

I currently can't access the hdmi port on the Jetson TX2 as it's mounted inside a robot, therefore I can't easily verify if the RealSense viewer is working on the TX2, I will check this when I run out of other options.

The version of JetPack I am using is 4.5.1.

My set-up was working until I updated the firmware on the RealSense to try and fix a freezing issue , unfortunately after I updated the firmware I got an opcode error with the version of the RealSense SDK I was using, so I upgraded the RealSense SDK and the ROS wrapper but now it doesn't work at all.

I would greatly appreciate any further thoughts or suggestions...

MartyG-RealSense, 2021-03-12:
Was your JetPack version updated recently when you were updating your RealSense SDK setup?  At the time of writing this, librealsense supports **JetPack 4.4**, with support for 4.4.1 upcoming.

https://github.com/IntelRealSense/librealsense/issues/8254#issuecomment-782033032

If there was a problem related to the JetPack version, I would expect that to manifest as a **Hunk** error, as described in https://github.com/IntelRealSense/librealsense/issues/8254 - have you experienced any of those please?

danielsmith185742, 2021-03-14:
I updated the version of JetPack when I updated the RealSense SDK. I am going to try and refresh my Jetson, change the version of Jetpack to v4.4 and reinstall everything to see if it helps.

Thxs

danielsmith185742, 2021-03-17:
Hi,

Just to inform you that after reflashing the TX2 with Jetpack v4.4 and rebuilding librealsense2 v2.42.0 all the errors disappeared. I have no idea exactly what was wrong before but everything is working now. Thanks for your assistance.

closed 2021-03-17 by danielsmith185742

MartyG-RealSense, 2021-03-17:
Excellent news @danielsmith185742 - thanks very much for the update!