realsenseai/librealsense #8548: Setting color sensor intrinsics

kogan-alex, 2021-03-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.06.00 | 
| Operating System & Version |  Linux (Ubuntu 18.04.1)  | 
| Kernel Version (Linux Only)    | 5.4.0                                       | 
| Platform                                 | PC  |
| SDK Version                            |  2.30.0.1216                     | 
| Language                            |  python }                          | 
| Segment			|  Robot                          | 

### Issue Description
We implemented our own hand-eye calibration procedure, as a by product we get calibrated camera intrinsics and we wish to save the results on the camera to have a more precise 3d/2d overlay (and for any purpose the intrisics are used). Is there a way to do this programmatically in Python?

MartyG-RealSense, 2021-03-10:
Hi @kogan-alex   I researched your question carefully and hope that the information below will be helpful.

In the case in the link below, a RealSense user sets values for the intrinsics by creating an intrinsics object, though it sounds as though you have accomplished that part of the project already.

https://github.com/IntelRealSense/librealsense/issues/4061

My understanding of the calibration writing process in Python is that **set_calibration_table()** is called first to set the current calibration to the dynamic area, and then **cal.write_calibration()** is called to save the new calibration permanently to the device's EEPROM storage.

https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.auto_calibrated_device.html

A Python script in the link below demonstrates these principles.

https://github.com/IntelRealSense/librealsense/issues/7087

[image]

kogan-alex, 2021-03-10:
Thanks for the quick reply, perhaps I explained myself poorly. We don't run Intel's "on chip calibration" but have our own method of obtaining: principle point, focal length and camera distortion coefficients. I don't think that the calibration table is written in a format that allows easy editing.

MartyG-RealSense, 2021-03-10:
I did understand that you were not using On-Chip Calibration.  There is not extensive documentation on the subject of developing your own calibration-writing mechanism outside of the On-Chip tool.  My interpretation (not certainty) of the documentation though is that the calibration commands mentioned above to store the current calibration values and write them to the device can be called independently from the On-Chip tool.

My further understanding is that the **CustomRW** tool that is part of the Dynamic Calibrator can output human-readable calibration information as an xml file, but the information stored in the calibration table is not in human-readable format.  This subject is covered in this discussion:

https://support.intelrealsense.com/hc/en-us/community/posts/360051257494-Custom-calibration-D410-and-D415-

kogan-alex, 2021-03-10:
I tried editing the intrinsics using the Realsense Viewer and noticed 2 issues:

* The values there are different from the output of the command:
`        intrinsics = cam_profile.get_stream(rs.stream.color).as_video_stream_profile().get_intrinsics()
`![Screenshot from 2021-03-10 18-19-44](https://user-images.githubusercontent.com/62833431/110661550-5a685d80-81cd-11eb-9072-aa81139491f0.png)

Which are correct for the color sensor?
* The "Write Table" button is grayed out. Is there another way of configuring the changes via RS viewer?

MartyG-RealSense, 2021-03-10:
It looks as though this window is being generated by the script in th elink below:

https://github.com/IntelRealSense/librealsense/blob/master/common/calibration-model.cpp

The script only references **depth** calibration, which is likely why it does not match up with your color intrinsics.

The **Write Table** button becomes clickable if you first enable the depth stream and **then** open the Camera Calibration window.

kogan-alex, 2021-03-11:
Thanks! The **CustomRW** tool does the job. 2 followup questions:
1. How can I specify the resolution corresponding to the desired intrinsics in the tool? (It only outputs information about 1920x1080)
2. This the only way? No support via Python API?

MartyG-RealSense, 2021-03-11:
2.  For developers who want to create their own custom calibration tool with the Calibration API, Intel's custom calibration white-paper document does contain examples of interfacing with the tool through OpenCV.  Given the close ties between Python and OpenCV, this may provide an indirect route for accessing the tool from Python.

Details of this programming interface begin on page 42 of the paper.

https://dev.intelrealsense.com/docs/d400-series-custom-calibration-white-paper

The Calibration API has an example program called **CustomCalibration** that is provided in the **examples** folder of the Custom Calibration tool package, and that example is described in this part of the paper.

Page 11 of the paper, meanwhile, has a table describing the formats supported by the Custom Calibration API.  The RGB section of this table only specifies support for 1920x1080 at 15 FPS.

[image]

kogan-alex, 2021-03-16:
Thank you for the detailed response. A related question: how sure are you in the camera intrinsics? What is the maximal error in focal point for example? I'm asking since we sometimes get widely different results in our proprietary calibration procedure.

MartyG-RealSense, 2021-03-16:
I do not have an answer that I can offer about the question of 'maximal error'.  It is not unusual though to get different results from a calibration.  This is why the SDK's calibration tools (Dynamic Calibrator and On-Chip) provide the option of dismissing a completed calibration and repeating the process before writing it to the camera once satisifed with the calibration results.

The On-Chip calibrator generates a health-check score after calibration and provides descriptive feedback about whether a result is good or could be improved.  This helps to guide the user's choice to write that calibration to the camera or repeat the process.

[image]

MartyG-RealSense, 2021-03-23:
Hi @kogan-alex  Do you require further assistance with this case, please?  Thanks!

kogan-alex, 2021-03-23:
Not for now. Thank you!

closed 2021-03-23 by kogan-alex