realsenseai/librealsense #8550: 3D Image Flashes then Disappears

JerryGadd, 2021-03-10:
|**librealsense**|2.41.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|f0244833|
|**Firmware Version**|01.05.03.00|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|0003ac79e069|
|**Firmware Update Id**|0003ac79e069|

Please provide a description of the problem
3D Image is displayed for a few seconds, then starts flashing off/on, and then disappears altogether!

https://user-images.githubusercontent.com/37139707/110655509-06527e80-81b7-11eb-8791-30cd11f8a085.mp4

https://user-images.githubusercontent.com/37139707/110678378-293c5d00-81ce-11eb-8a50-af619de28c00.mp4

itSeez, 2021-03-17:
I have the same issue. Try disabling both of the GLSL settings in the RealSense viewer.

RealSenseSupport, 2021-03-26:
@JerryGadd 

Are you still seeing this issue?  Can you provide the Viewer settings that you have enabled?

Have not been able to reproduce this with the latest 2.43 Viewer release.  Were you able to give the latest Viewer version a try?

Nir-Az, 2021-03-30:
Hi @JerryGadd , Can you please confirm that removing the "Use GLSL for processing" checkbox fix it?
Can you also provide the system details you run it on? (Laptop? Model? Graphic card?)

Thanks

JerryGadd, 2021-04-06:
Yes it did fix it, until the camera stopped working completely several days later!

  *   Laptop
  *   HP ZBook 17 G4
  *   Nvidia Quadro P3000

Siemens Industry Software Limited is a limited company registered in England and Wales.
Registered number: 3476850.
Registered office: Faraday House, Sir William Siemens Square, Frimley, Surrey, GU16 8QD.

RealSenseSupport, 2021-04-14:
Thanks for the information

closed 2021-04-14 by RealSenseSupport