realsenseai/librealsense #8551: How can I just get the data from realsense device on android : depth sensor, rgb camera .

bjain409, 2021-03-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2021-03-11:
Hi @bjain409   The librealsense Android wrapper's example program **capture** may meet your needs with some modification to disable visual rendering, as it makes use of depth and RGB streams.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/capture/

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/capture/src/main/java/com/intel/realsense/capture/MainActivity.java#L163

My understanding is that the visual rendering is handled by the OpenGL **GLRsSurfaceView** elements of the script.  So you could try removing or commenting-out these parts out or removing them until the program meets your needs of displayless streaming.

bjain409, 2021-03-12:
> Hi @bjain409 The librealsense Android wrapper's example program **capture** may meet your needs with some modification to disable visual rendering, as it makes use of depth and RGB streams.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/capture/
> 
> https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/capture/src/main/java/com/intel/realsense/capture/MainActivity.java#L163
> 
> My understanding is that the visual rendering is handled by the OpenGL **GLRsSurfaceView** elements of the script. So you could try removing or commenting-out these parts out or removing them until the program meets your needs of displayless streaming.
> 
> The case in the link below about the Android **capture** example includes an image of the unmodified program running on a phone, to provide you with an illustration of the captured data:
> 
> [#4091 (comment)](https://github.com/IntelRealSense/librealsense/issues/4091#issuecomment-502467123)
> 
> [image]
Hi,
@MartyG-RealSense 
Thanks for responding
I am not completely following the process.
Let me tell you the actual requirement.
I am building an android apk with realsense 4351i.
wherein I will create an trigger from android end  for capture.
The apk will be a  background process running continously .
After capture is triggered I need object of rgb camera and depth camera. is there any API defined . which returns be just the object of bitmaps.
If not can you please create a service which returns. 
.

Thanks..

bjain409, 2021-03-12:
> Hi @bjain409 The librealsense Android wrapper's example program **capture** may meet your needs with some modification to disable visual rendering, as it makes use of depth and RGB streams. [...]
@MartyG-RealSense 
Can you please create a service which will return the rgb and dept camera image
which accepts an flag 2d or 3d(optional if not feasible for android framework) and returns result to calling thread?

MartyG-RealSense, 2021-03-12:
My understanding from your description earlier is that the Android app will be streaming depth and color continuously.  You wish to be able to trigger a capture, which will save depth and color images.  Is that correct, please?

If it is, the links below provide examples of scripting references for converting an Android frame to a bitmap and then saving it as an image.  These could possibly be used to program the desired feature in your Android application yourself.

https://github.com/IntelRealSense/librealsense/issues/5446

https://github.com/IntelRealSense/librealsense/issues/4455

https://github.com/IntelRealSense/librealsense/issues/6089

https://github.com/IntelRealSense/librealsense/issues/8316

bjain409, 2021-03-19:
> My understanding from your description earlier is that the Android app will be streaming depth and color continuously. You wish to be able to trigger a capture, which will save depth and color [...]

MartyG-RealSense, 2021-03-19:
Hi @bjain409   Your new comment above just quotes my previous comment from 7 days ago.  Do you require further assistance please?  Thanks!

bjain409, 2021-03-19:
> Hi @bjain409 Your new comment above just quotes my previous comment from 7 days ago. Do you require further assistance please? Thanks!

@MartyG-RealSense  I am not getting proper support for my requirement.
@dorodnic  Can you please help me with my requirements?

Thanks.

MartyG-RealSense, 2021-03-19:
Dorodnic is not available.

It is important to be realistic with expectations.  In most cases, a new feature cannot be created immediately.  We can certainly provide developers with advice about scripting as I have done earlier in this discussion, but the emphasis is on the customer using that advice to develop their own solution.

MartyG-RealSense, 2021-03-25:
Hi again @bjain409  Is there anything more that I can do for you, bearing in mind that Intel cannot quickly create an Android feature to meet your needs.

bjain409, 2021-03-25:
> Hi again @bjain409 Is there anything more that I can do for you, bearing in mind that Intel cannot quickly create an Android feature to meet your needs.

@MartyG-RealSense 
By when I can expect this feature?

MartyG-RealSense, 2021-03-25:
A **feature request** can be made but the process can take a significant period of time to complete and does not necesarily result in the feature request being approved for creation.  This is why the preferable option is to develop your own solution based upon advice provided on this forum.

bjain409, 2021-03-26:
> A **feature request** can be made but the process can take a significant period of time to complete and does not necessarily result in the feature request being approved for creation. This is why the preferable option is to develop your own solution based upon advice provided on this forum.

@MartyG-RealSense 
None of the solutions provided is useful.

MartyG-RealSense, 2021-03-26:
@bjain409   In your opening message of this discussion, you stated the requirement for depth and RGB data on Android but with no UI rendering.  As the Android **Capture** example provides simultaneous depth and RGB streaming, I suggested using that example's code but disabling the visual rendering elements so that your application could access the live stream data without needing to render it on-screen.

After that suggestion, you added the following requirement:

"I am building an Android apk with RealSense 4351i, wherein I will create an trigger from Android end for capture.  The apk will be a background process running continuously.  After capture is triggered I need object of RGB camera and depth camera.  Is there any API defined which returns just the object of bitmaps".

The **Capture** code with rendering disabled would seem to meet your needs for a background process running continuously.  Am I correct in thinking that you also require a trigger mechanism in the Android app that will export a depth image and RGB image of the frame that was captured when the trigger was activated, please?  And if that is correct, would the trigger be automatic (for examle, a capture every 5 minutes) or would it be manually activated by a button press on the app?

In link https://github.com/IntelRealSense/librealsense/issues/8551#issuecomment-797663275 I pointed to examples of code for converting an Android frame to a bitmap and capturing it as a image.

If that code is suitable for creating an image capture mechanism, then the only part of the project that would seem to be missing is the trigger mechanism.

If the suggestions made in this case have not been suitable for your needs, could you provide information please about why it is not suitable for your project so that I can research alternative approaches.  Thanks very much.

MartyG-RealSense, 2021-04-01:
Hi @bjain409   Could you provide feedback about the advice provided in the comment above please?  Thanks!

bjain409, 2021-04-01:
> Hi @bjain409 Could you provide feedback about the advice provided in the comment above please? Thanks!

@MartyG-RealSense 
I am not at all satisfied with the advice.
I thought purchasing intel-realsense device would be greatly helpful and would have support for it, But it's not true!
I might get the depth sensor integrated with RGB of different OEMs.
At least that would not cost me as high as intel costs me.

MartyG-RealSense, 2021-04-01:
I am sorry that you feel that your needs have not been met.  Extensive advice about programming an Android application to fulfill the functions that you require has been provided in the comments above.  There is not any more that I can do to help in this particular case.

bjain409, 2021-04-01:
@MartyG-RealSense Can you check 
It's not working adding a dependency
as in
"https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/java_example/readme.md"

MartyG-RealSense, 2021-04-01:
It looks as though in this example, the dependency code mentioned in the above link is located in the **build.gradle** file in the folder **examples/java_example/app/**

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/java_example/app/build.gradle#L30

[image]

bjain409, 2021-04-02:
> It looks as though in this example, the dependency code mentioned in the above link is located in the **build.gradle** file in the folder **examples/java_example/app/**
> 
> https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/java_example/app/build.gradle#L30
> 
> [image]

Hello @MartyG-RealSense 
I have followed the same instruction
But while attaching a device  get the pop-up only once
Not like in the case of the capture or recording module wherein I get the pop-up twice.
So the app crashes and I don't get frames

bjain409, 2021-04-02:
> Hello @MartyG-RealSense
> I have followed the same instruction
> But while attaching a device get the pop-up only once
> Not like in the case of the capture or recording module wherein I get the pop-up [...]

@MartyG-RealSense 
https://github.com/bjain409/libRealsenseViaDependency.git
This is what I did. Its not working

MartyG-RealSense, 2021-04-02:
It looks as though the Android example has been cloned to your GitHub so it is mostly the same except for the **MainActivity**.java file.

Did the program run okay on your GitHub in its original form before you began adapting it?

bjain409, 2021-04-02:
> It looks as though the Android example has been cloned to your GitHub so it is mostly the same except for the **MainActivity.java** file.
> 
> Did the program run okay on your GitHub in its original form before you began adapting MainActivity.java?
@MartyG-RealSense 
It's not working. I copied the same content from the capture module and in the java example as you can see.
Did you try running?

bjain409, 2021-04-02:
> It looks as though the Android example has been cloned to your GitHub so it is mostly the same except for the **MainActivity.java** file.
> Did the program run okay on your GitHub in its original form [...]
@MartyG-RealSense 
I had invested more than 10 days in using dependencies 
But it was not helpful. I was working with the whole project I was using the capture module. I thought something is wrong at my end. But as you suggested I followed the same. Its still not helpful
Can you plz check it

MartyG-RealSense, 2021-04-02:
My Android programming knowledge is limited, so I will seek advice from Intel about your case.

bjain409, 2021-04-02:
> My Android programming knowledge is limited, so I will seek advice from Intel about your case.

@But were you able to run the app?

MartyG-RealSense, 2021-04-02:
I am not able to test Android programs myself unfortunately.  I have sent a message to Intel for programming advice about your case.

bjain409, 2021-04-03:
> I am not able to test Android programs myself unfortunately. I have sent a message to Intel for programming advice about your case.

@MartyG-RealSense 
Please resolve this asap.
This is blocking my further development.

MartyG-RealSense, 2021-04-03:
I have talked about your case with Intel.  Another RealSense team member will join this discussion.

bjain409, 2021-04-03:
> I have talked about your case with Intel. Another RealSense team member will join this discussion.

@MartyG-RealSense 
Just a question I want to ask.
How can something get merged into master without being verified

MartyG-RealSense, 2021-04-03:
If you are referring to Pull Requests, then you could edit a change into the source code version of librealsense and build it.  Because it is a private custom version of librealsense though, only your librealsense build would have it and the change would not be a part of the public master branch of librealsense.

bjain409, 2021-04-03:
> If you are referring to Pull Requests, then you could edit a change into the source code version of librealsense and build it. Because it is a private custom version of librealsense though, only your librealsense build would have it and the change would not be a part of the public master branch of librealsense.
> 
> Some RealSense users make a custom version by cloning a **fork** of librealsense from the librealsense GitHub to their own GitHub and then using that version.

@MartyG-RealSense 
Ok, nevermind please fix the issue asap.

MartyG-RealSense, 2021-04-03:
The other RealSense team member will work with you on your case when they join it.  Please work with them.  Thanks very much.

bjain409, 2021-04-03:
> The other RealSense team member will work with you on your case when they join it. Please work with them. Thanks very much.

@MartyG-RealSense 
Can I expect their correspondence by eod?

MartyG-RealSense, 2021-04-03:
As it is Saturday at the time of writing this, it may be after the weekend now.

bjain409, 2021-04-03:
> As it is Saturday at the time of writing this, it may be after the weekend now.

@MartyG-RealSense 
Ok no worries
Just ask them to correspond asap.
Thanks for being kind and being helpful🙏

bjain409, 2021-04-06:
> > As it is Saturday at the time of writing this, it may be after the weekend now.
> 
> @MartyG-RealSense
> Ok no worries
> Just ask them to correspond asap.
> Thanks for being kind and being helpful

@MartyG-RealSense 
Any update?

MartyG-RealSense, 2021-04-06:
I will send Intel a reminder now.

RealSenseSupport, 2021-04-07:
@bjain409 Sorry I didn't follow your requirement. If for the requirement in the ticket title, I think MartyG has replied the good solution. https://github.com/IntelRealSense/librealsense/issues/8551#issuecomment-808031352  Or could you please clarify more about your requirements? Or anything missing in MartyG's reply? Thanks!

And for your issue mentioned later, I saw you created another ticket https://github.com/IntelRealSense/librealsense/issues/8728 We'll look into this issue in that ticket.

bjain409, 2021-04-07:
> @bjain409 Sorry I didn't follow your requirement. If for the requirement in the ticket title, I think MartyG has replied the good solution. [#8551 (comment)](https://github.com/IntelRealSense/librealsense/issues/8551#issuecomment-808031352) Or could you please clarify more about your requirements? Or anything missing in MartyG's reply? Thanks!
> 
> And for your issue mentioned later, I saw you created another ticket #8728 We'll look into the later issue in that ticket.

@RealSenseSupport 
Thanks for responding.
The current requirement is #8728 Sorry I quoted that issue here.
Can you please fix issue #8728 asap.
Currently, I am directly working on the capture module And I can't delete any of the other folders from root As it breaks the app.
By when it can be done?

RealSenseSupport, 2021-04-07:
@bjain409 So can I understand this issue https://github.com/IntelRealSense/librealsense/issues/8551 resolved now? 

For https://github.com/IntelRealSense/librealsense/issues/8728, we need more information from you to debug. Please move to that issue for more details. Thanks!

bjain409, 2021-04-07:
> @bjain409 So can I understand this issue #8551 resolved now?
> 
> For #8728, we need more information from you to debug. Please move to that issue for more details. Thanks!

#8551 is still not done.
What I need is not provided.

RealSenseSupport, 2021-04-07:
@bjain409 Okay. Please clarify you detailed requirement in this ticket #8551.

bjain409, 2021-04-07:
> @bjain409 Okay. Please clarify you detailed requirement in this ticket #8551.

@RealSenseSupport 
I am building an android app that will be used for object detection.
It will background app. It won't have any UI components.
There will be a job that will trigger a capture event. In capture, it will store RGB image and depth sensor data.
For instance, take an example there will be a job which will run every 30mins.
Now every 30min images should be captured from both RGB and depth sensors simultaneously.
So I need a method that will return me RGB data and I need a matrix of depth data.
In the case of the depth sensor, I need the depth of individual pixel/point from the depth sensor.
I should get all the depth data from the individual point. It's required for 3D imaging to verify whether the object is real or just the 2d image/photo.
For instance, Consider I have capture triggered and now I want to recognize humans.
So by default nose tip will have the shortest distance from the sensor and the eyes will have more distance.
This will guarantee no one is spoofing. This is just the use case.

bjain409, 2021-04-09:
> @RealSenseSupport
> I am building an android app that will be used for object detection.
> It will background app. It won't have any UI components.
> There will be a job that will trigger a capture event. [...]

@RealSenseSupport 
Any update?

RealSenseSupport, 2021-04-12:
@bjain409 I think MartyG has provided some reference to you in this ticket. Could you please check if these examples help you ? https://github.com/IntelRealSense/librealsense/issues/8551#issuecomment-797663275

bjain409, 2021-04-15:
@MartyG-RealSense 
After getting the depth frame and video frame.
We have a method frame.getHeight,  getWidth
are these values constants?
Does it change in any case at runtime?

MartyG-RealSense, 2021-04-15:
Hi @bjain409  Your case is being handled by @RealSenseSupport now.  Thank you.

RealSenseSupport, 2021-04-20:
@bjain409 The value got from frame.getHeight, getWidth are constant which are the resolution you configure or default value if not configured. They will not change during streaming.

RealSenseSupport, 2021-04-27:
@bjain409 Any other questions for this ticket? Thanks for your update!

RealSenseSupport, 2021-05-12:
@bjain409 Anything else we can help on this ticket? Looking forward to your update. Please note that this ticket will be closed if we don't hear from you for another 7 days. Thanks!

closed 2021-05-17 by bjain409