realsenseai/librealsense #8554: The built version in windows cannot find a device

seliber, 2021-03-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (05.12.10.00) | 
| Operating System & Version |   {Win 10  | 
| SDK Version                            | 2.41.0                          | 
| Language                            |  {C/C#}                          | 

### Issue Description
I built up the source successfully in windows, but the builted "realsense-viewer.exe", cannot found any devices after launch. 
If I ran some sample such as: cs-tutorial-1-depth.exe, also get the result "There are 0 connected RealSense devices.".
After doing some debug, I found there was no item in registry which could fill the regex _regex_camera_interface_ in _enumerator-winusb.cpp_

---
But when I launched the version "RealSense-viewer.exe" installed from setup file _Intel.RealSense.SDK-WIN10-2.41.0.2657.exe_ , the device could be recognized and worked well. 

---
Another question is:
the size of "realsense2.dll" installed is about 30M, but the dll built from source is only about 7~8M. Is there anything I miss to build the source?

`cmake ../. -DBUILD_NETWORK_DEVICE=ON -DBUILD_CSHARP_BINDINGS `
---
Is there anything I need to config? Or what can I do to proceed with the version I built from source?

MartyG-RealSense, 2021-03-11:
Hi @seliber   You can reduce the size of a librealsense build by configuring CMake to "strip" out debug symbols and other content.

https://github.com/IntelRealSense/librealsense/issues/7982

In regard to your build of librealsense for Windows from source, your CMake instruction is a little minimal.  I would also include the additional build flags below to build in the optimized Release mode instead of Debug mode, and include support for examples and tools (including the RealSense Viewer).

**DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

If you are building with CMake-GUI, the Windows source-build instructions provide an image with ticked flags that you can replicate.

[image]

seliber, 2021-03-12:
Dear @MartyG-RealSense , here is my gui for build the source, it's almost the same with the image, except few examples. 
[image]

But my key issue is : the built version cannot find a device. Do you think the build of examples will impact the result?

seliber, 2021-03-12:
After build with BUILD_EXAMPLES, it can find the device then. 
Close the issue, but it is really strange....

closed 2021-03-12 by seliber

MartyG-RealSense, 2021-03-12:
Excellent to hear that using **BUILD_EXAMPLES** worked for you.   Thanks for the update!