realsenseai/librealsense #8556: Fatal Error on previsouly working code

incinemator, 2021-03-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415  | 
| Firmware Version                   | 05.12.11.00 | 
| Operating System & Version |   Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  5.4.0-66-generic  | 
| Platform                                 | PC |
| SDK Version                            |   2.0  | 
| Language                            |  C/C++  |  

### Issue Description
Hello all,

I am trying to run some sample codes but I run into errors on code that was previously working. Most commonly I get the following error: 
fatal error: ../example.h: No such file or directory
 #include "../example.h"

Any suggestions?

MartyG-RealSense, 2021-03-11:
Hi @incinemator   The error **#include "../example.h"** makes me think that you are having problems with the "C" examples in particular, such as **rs-color.c** and **rs-distance.c**.  Is that correct, please?  

If it is C examples that you are experiencing problems with, are you having any problems with **C++** examples?

incinemator, 2021-03-11:
Hi @MartyG-RealSense, yes indeed I'm having problems both with C and C++ examples.

MartyG-RealSense, 2021-03-11:
Thank you for the confirmation.  Have you changed your librealsense SDK version recently?  If you have, which version number were you using before and which version are you using now (for example, **2.42.0**).

Also, what errors do you receive from the C++ examples?

incinemator, 2021-03-11:
Than you for your reply. I didn't change version. This is the output of rs-capure:

rs-capture.cpp:2:10: fatal error: example.hpp: No such file or directory
 #include "example.hpp"          // Include a short list of convenience functions for rendering
          ^~~~~~~~~~~~~
compilation terminated.

And this is of rs-hello :

rs-hello-realsense.cpp:7:6: error: ‘Pipeline’ in namespace ‘rs2’ does not name a type
 rs2::Pipeline p;
      ^~~~~~~~
rs-hello-realsense.cpp:10:1: error: ‘p’ does not name a type
 p.start();
 ^
rs-hello-realsense.cpp:13:6: error: ‘framese’ in namespace ‘rs2’ does not name a type
 rs2::framese frames = p.wait_for_frames();
      ^~~~~~~
rs-hello-realsense.cpp:16:26: error: ‘frames’ was not declared in this scope
 rs2::depth_frame depth = frames.get_depth_frame():
                          ^~~~~~
rs-hello-realsense.cpp:16:26: note: suggested alternative: ‘free’
 rs2::depth_frame depth = frames.get_depth_frame():
                          ^~~~~~
                          free
rs-hello-realsense.cpp:23:43: error: ‘width’ was not declared in this scope
 float dist_to_center = depth.get_distance(width / 2, height / 2);
                                           ^~~~~
rs-hello-realsense.cpp:23:43: note: suggested alternative: ‘wcwidth’
 float dist_to_center = depth.get_distance(width / 2, height / 2);
                                           ^~~~~
                                           wcwidth
rs-hello-realsense.cpp:26:1: error: ‘cout’ does not name a type
 cout << "The camera is facing an object " << dist_to_center << " meters away \r";
 ^~~~
painkiller@painkiller-Lenovo-G500s:~/Documents/RealSense$ g++ -std=c++11 rs-capture.cpp -lrealsense2
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/Scrt1.o: In function `_start':
(.text+0x20): undefined reference to `main'
collect2: error: ld returned 1 exit status

incinemator, 2021-03-11:
Nevermind the last 4 lines!

incinemator, 2021-03-13:
Ok, Thanks for the help!

incinemator, 2021-03-18:
Hi again, I reinstalled librealsense but the problem remained. I even did a clean install of ubuntu and switched to 20.04, installed librealsense again and I get the same errors... I noticed that while running the realsense viewer I get several errors:

 18/03 23:57:50,207 WARNING [139748662482688] (backend-v4l2.cpp:1357) Pixel format 20303157-0000-0010-8000-00aa003 likely requires patch for fourcc code W10 !
 18/03 23:57:50,228 WARNING [139748662482688] (backend-v4l2.cpp:1357) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 18/03 23:57:50,242 WARNING [139748662482688] (sensor.cpp:1086) RGB Camera: skipping Backlight Compensation control. descriptor: [min/max/step/default]= [0/0/0/0]
 18/03 23:57:50,243 WARNING [139748662482688] (sensor.cpp:1086) RGB Camera: skipping Exposure control. descriptor: [min/max/step/default]= [0/0/0/0]
 18/03 23:57:50,244 ERROR [139748662482688] (types.h:315) xioctl(VIDIOC_G_CTRL) failed Last Error: Invalid argument
 18/03 23:57:50,244 WARNING [139748662482688] (sensor.cpp:1112) Failed to add Enable Auto Exposure control for RGB Camera
 18/03 23:57:55,932 WARNING [139748395951872] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 18/03 23:57:55,932 WARNING [139748395951872] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 18/03 23:57:55,933 WARNING [139748395951872] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 18/03 23:57:55,933 WARNING [139748395951872] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 18/03 23:57:56,701 WARNING [139748387559168] (hdr-merge.cpp:38) HDR Merge filter cannot process frames because relevant metadata params are missing
 18/03 23:57:56,764 WARNING [139748077197056] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 18/03 23:57:56,764 WARNING [139748077197056] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 18/03 23:57:56,764 WARNING [139748077197056] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 18/03 23:57:56,765 WARNING [139748077197056] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 18/03 23:57:58,592 ERROR [139748077197056] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!

MartyG-RealSense, 2021-03-19:
There may be something on your computer that librealsense is conflicting with, whether it is the Linux version, kernel version or both.  It is unusual, considering that you have stated that the code has worked previously (though this could still mean that the code is fine and the problem is Linux-related).

To eliminate the possibility that such a conflict is occurring, would it be possible to try building librealsense with CMake using the **RSUSB backend** method if you have not done so already?  This installation method is not dependent on Linux versions or kernel versions and does not require patching.  Instructions for the RSUSB method that uses the CMake build flag **-DFORCE_RSUSB_BACKEND=true** are in the link below.

https://github.com/IntelRealSense/librealsense/issues/6432#issuecomment-631484711

MartyG-RealSense, 2021-03-25:
Hi @incinemator  Do you require further assistance with this case, please?  Thanks!

incinemator, 2021-03-25:
Hi @MartyG-RealSense , sorry I forgot to reply. I haven't tried the above solution yet, if it doesn't work out again I will let you know. Thanks!

MartyG-RealSense, 2021-03-26:
No problem @incinemator - thanks very much for the update.  Good luck!

MartyG-RealSense, 2021-04-07:
Hi @incinemator  Do you still have a problem that you require assistance with, please?  Thanks!

incinemator, 2021-04-08:
Hi @MartyG-RealSense I attempted to try the method you mentioned but I could not find the root folder of realsense. I uninstalled the package and try to build it from source

MartyG-RealSense, 2021-04-08:
Consider the **root** directory of the source code folder of librealsense (the one downloaded from the **Assets** file list) to be the one with the same set of folders as the librealsense GitHub front page.  In the unzipped source-code folder, the root directory should be in **librealsense > librealsense**:

[image]

incinemator, 2021-04-18:
Hi @MartyG-RealSense  sorry for the delayed answer. I think the problem is the kernel version, for now I will try to build it from source and see what happens.. Thanks!

MartyG-RealSense, 2021-04-19:
Thanks very much @incinemator for the update.  I look forward to your test results.

MartyG-RealSense, 2021-05-02:
Hi @incinemator  Do you have an update about this case that you can provide, please?  Thanks!

incinemator, 2021-05-06:
Hi @MartyG-RealSense, again sorry for the late response,  I tried building realsense manually but encountered the same problem... Moreover when I run realsense-viewer it asks me to upgrade the firmware and while upgrading, at 305 a disconnected camera error apperas...

MartyG-RealSense, 2021-05-06:
If you mean "30%", that is the point in the Viewer's firmware update process when the camera disappears from the side-panel of the Viewer as it switches to **recovery mode** and the firmware begins being updated.  When the 100% point in the installation is reached, the camera re-appears in the side-panel of the Viewer.

[image]

I can recall a couple of past cases though where the camera does not complete the transition to recovery mode after the 30% point, halting the firmware update.

Would it be possible to try updating the firmware with the SDK's other firmware updater tool **rs-fw-update** instead please?

https://github.com/IntelRealSense/librealsense/tree/master/tools/fw-update

incinemator, 2021-05-07:
I tried rs-fw-update but I get an error:
RealSense error calling ():
    failed to read firmware file

MartyG-RealSense, 2021-05-07:
Do you have the camera attached to a USB 3 port (if you have one available on your computer / computing device) when using rs-fw-update, please?

incinemator, 2021-05-07:
Yes

MartyG-RealSense, 2021-05-07:
Although you are unable to update the firmware, is the camera still detected in the RealSense Viewer and does the Viewer appear to work normally when streams are enabled?

incinemator, 2021-05-07:
Yes indeed, in the viewer it works normally

MartyG-RealSense, 2021-05-07:
Assuming that the SDK id working okay despte the firmware issue, I went back to the beginning of the case and the errors that you were getting in https://github.com/IntelRealSense/librealsense/issues/8556#issuecomment-802339730

They do look like kernel-related conflicts.  I suggested installing librealsense from source using the **RSUSB** installation method, which requires an internet connection but is not dependent on Linux versions or kernel versions and does not require patching.   Did you manage to build librealsense using the RSUSB method, please?

incinemator, 2021-05-07:
I tried it still no luck...

MartyG-RealSense, 2021-05-07:
Do you have a link to the instructions that you used to perform the RSUSB installation method, please?

incinemator, 2021-05-07:
I used the ones you provided, from another topic #6432 (comment)

MartyG-RealSense, 2021-05-07:
And the Viewer seemed to work fine on the RSUSB build but you still experienced the original problem with **example.h** when trying to run the example programs?

How are you running the example programs?  Are you launching them by inputting their names into the Ubuntu terminal?  For example:

**rs-align**
**rs-capture**
**rs-hello-realsense**

incinemator, 2021-05-07:
Yes I noticed the example programs have been built, I used g++ -std=c++11 filename.cpp -lrealsense2 according to installation guide here: https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

MartyG-RealSense, 2021-05-08:
The problem may be with using **g++ -std=c++11 filename.cpp -lrealsense2** with the SDK examples.  Whilst this may be suitable for an application that you have written yourself, the SDK example programs are built in a different way.

After the SDK examples have been built with CMake the first time with **-DBUILD_EXAMPLES=true** during installation of librealsense, they can be buitl again by repeating the **make** procedure.

https://github.com/IntelRealSense/librealsense/issues/2946#issuecomment-449290237

A RealSense user shared in the link below a guide that they wrote for compiling the examples with CMake on Ubuntu.

https://github.com/IntelRealSense/librealsense/issues/6797

MartyG-RealSense, 2021-05-11:
At this point, I believe that a fresh installation of librealsense with the distribution method that you previously used would be the best way forward.  This should eliminate the possibility that the issues in this case are due to something that has gone wrong in your current librealsense build.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

incinemator, 2021-05-14:
Hi @MartyG-RealSense I tried uninstalling but I get the response that librealsense2 is not installed.. I even tried cmake uninstalled && make clean but again the system replies that there's nothing to be removed. Is there a possibility that simply removing the librealsense folder would work? thanks

MartyG-RealSense, 2021-05-14:
If you have used **dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge** to remove all librealsense Debian packages and have also performed **sudo make uninstall && make clean** to clean up your source code build then you have taken all reasonable steps to clean up your previous librealsense installations.

If you still have a librealsense folder then you could delete it and start a new build.  I would recommend choosing one method or the other (Debian packages or source-code) but not attempting both.  This is because using both methods could result in a warning about having multiple udev files installed.

incinemator, 2021-05-20:
Hi @MartyG-RealSense . I tried the clean install but the problem is still there... Could it be possibly related to the nvidia graphics card driver? There are issues in general with newer ubuntu versions and legacy nvidia drivers. I think the problem appeared after installing a later version of the driver.

MartyG-RealSense, 2021-05-20:
I note that in comment https://github.com/IntelRealSense/librealsense/issues/8556#issuecomment-797123350 you mentioned that you had not changed your librealsense version.  Which librealsense version are you currently using please?

I ask this because from firmware version **5.12.8.200** it has been a requirement to use librealsense version **2.39.0** or newer due to internal changes in the firmware.  The firmware **5.12.11.0** firmware that you listed as your currently installed firmware at the top of this case is three versions newer than 5.12.8.200 and so would fall within the 'SDK 2.39.0 minimum' rule.

[image]

MartyG-RealSense, 2021-05-27:
Hi @incinemator  Do you have an update that you can provide about this case, please?  Thanks!

incinemator, 2021-06-02:
Hi @MartyG-RealSense sorry for not responding. I have no update for now but I have installed the latest version I think

MartyG-RealSense, 2021-06-03:
I read back over this case from the beginning.  In the comment below, you mention building librealsense examples using **g++ -std=c++11**

https://github.com/IntelRealSense/librealsense/issues/8556#issuecomment-834649766

You should not need to install the optional **librealsense2-dev** package to use the SDK examples, just the two packages 'librealsense2-dkms' and 'librealsense2-utils'. Installing **librealsense2-utils** will build the examples and tools, and then you can find them already built and ready to launch in the **usr/local/bin** folder of Ubuntu.

incinemator, 2021-06-08:
Hi @MartyG-RealSense I did as you suggested and finally the examples are working, thanks a lot for your time and help!

MartyG-RealSense, 2021-06-08:
That's great to hear @incinemator - you are very welcome.  Thanks for the update   :)

closed 2021-06-14 by MartyG-RealSense