realsenseai/librealsense #8557: The range of pixel value when using Y8 and Y16 format.

fwf-lernen, 2021-03-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.11.00 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0  | 
| Platform                                 | PC |
| SDK Version                            |  2.42.0  | 
| Language                            |  C   | 

### Issue Description
When I use infrared camera to capture images in Y8 format, no matter how I reduce exposure time(neither using realsense-viewer nor call API: sensor.set_option(RS2_OPTION_EXPOSURE, exposure_time) ), I can't get pixel value below 15 even though I set the exposure time to 1usec.
The same thing happens when using RGB camera to capture images in Y16 format (min pixel value is 16\*256 ). Besides, no matter how I increase exposure time, I can't get pixel value above 235\*256.
Is the pixel value range set to 16 \~ 255 in Y8 and 16\*256 \~ 235\*256 in Y16? If not, how can I get pixel value that exceed the limit?
[image]

MartyG-RealSense, 2021-03-12:
Hi @fwf-lernen   I have researched your question carefully.  Before moving forward though, could you provide further information about your meaning when you talk about the 'pixel value'?  

Also, can you explain what you are trying to achieve that requires the setting that you are aiming for please?  Thanks very much.

fwf-lernen, 2021-03-12:
Hi @MartyG-RealSense, thank you for your reply.
`
// include the librealsense C++ header file
#include <librealsense2/rs.hpp>

// include OpenCV header file
#include <opencv2/opencv.hpp>

using namespace std;
using namespace cv;

int main()
{
    //Contruct a pipeline which abstracts the device
    rs2::pipeline pipe;

    //Create a configuration for configuring the pipeline with a non default profile
    rs2::config cfg;

    //Add desired streams to configuration
    cfg.enable_stream(RS2_STREAM_INFRARED, 640, 480, RS2_FORMAT_Y8, 30);

    //Instruct pipeline to start streaming with the requested configuration
    pipe.start(cfg);

    // Camera warmup - dropping several first frames to let auto-exposure stabilize
    rs2::frameset frames;
    for(int i = 0; i < 30; i++)
    {
        //Wait for all configured streams to produce a frame
        frames = pipe.wait_for_frames();
    }

    //Get each frame
    rs2::frame ir_frame = frames.first(RS2_STREAM_INFRARED);

    // Creating OpenCV matrix from IR image
    Mat ir(Size(640, 480), CV_8UC1, (void*)ir_frame.get_data(), Mat::AUTO_STEP);
    imwrite("ir_image.png", ir);

    return 0;
}

`
Use this code to capture a image in Y8 format with a infrared camera , then just use " cv::Mat img = cv::imread("ir_image.png", IMREAD_GRAYSCALE) " to read the image. All values of the Mat are above 15, but I want to get values below 15. However, I can't do this by decreasing the exposure time.

MartyG-RealSense, 2021-03-12:
Thanks very much for the information.  I thought that you may have been talking about a cv::mat when you mentioned a range of 0 to 255 but wanted to make sure.  I appreciate the confirmation.

The link below describes how OpenCV has a mechanism that can change the pixel value to a custom-defined value if it is less or greater than a certain amount.  So for example, if the pixel value =16 then that code could set that pixel to a lower value such as 1 or 0.  Would that be a practical solution for you?

https://stackoverflow.com/questions/37342716/set-every-pixel-of-a-mat-to-a-certain-value-if-it%C2%B4s-lower-than-a-value

fwf-lernen, 2021-03-12:
Thanks very much for your suggestion. But what I actually want to do is to calibrate the camera's  inverse response function. So if I just set the value below 17 to 1 or 0, that means I can't calibrate the pixel value between 0 and 16. Honestly, if I can't get pixel value below 16, that means I needn't calibrate these values. I am here to find out why I can't get values below 16 directly. Is the camera set to be like this?

MartyG-RealSense, 2021-03-12:
I ran the testing again in the Viewer in infrared mode with exposure at '1' and observed that even with **gain** minimized at the lowest allowed value of 16, detail could be faintly seen in the darkened image as the camera is moved around.  This suggests to me that as long as gain cannot be reduced below '16' then the '16' values in your results may represent the faintly visible detail on the IR image.

fwf-lernen, 2021-03-12:
OK, I think I have got what I want. Thanks again!
By the way, should I close the issue or you will do it? If you can, please close it. (I am going to sleep and can't reply to you in time.)

MartyG-RealSense, 2021-03-12:
You are very welcome.   :)   I can close the case for you.

closed 2021-03-12 by MartyG-RealSense