realsenseai/librealsense #8562: L515 Error on second start

bluebass44, 2021-03-12:
Camera Model                       L515 
Firmware Version                   01.05.04.01
Operating System & Version Windows 10
Language                              C++

This generates an error the **second** time you run it after unplugging the camera from the PC.
```
rs2::config cfg;
cfg.disable_stream(RS2_STREAM_ANY);
cfg.enable_stream(RS2_STREAM_COLOR, 1920, 1080, RS2_FORMAT_RGB8, 30); 
cfg.enable_stream(RS2_STREAM_DEPTH, 1024, 768, RS2_FORMAT_Z16, 30);  // <---Remove this

rs2::pipeline p;
p.start(cfg);
```

If you remove the depth stream it does not generate an error no matter how many times you run it. 
What am I doing wrong?

RealSenseSupport, 2021-03-26:
Hi @bluebass44 

Are you still seeing issues with this?

bluebass44, 2021-03-26:
Yes.

RealSenseSupport, 2021-04-05:
Can you provide some more details on the error and exact steps to reproduce so we can attempt to on our end? 

Is this code the only thing in your source file? 

And what do you mean by second time? 

Thanks

bluebass44, 2021-04-06:
Hi,

I discovered there is a setting to turn off the sensors ability to
disregard certain pixels based on their accuracy.  This is what was causing
the dark pixels.  Is there any negative impact on the accuracy of future
frames?

Regards,
Micah

RealSenseSupport, 2021-04-12:
Which specific setting are you referring to? Are the dark pixels what you meant by "error" in your original comment? And what do you mean by negative impact?

Also if you could answer these questions https://github.com/IntelRealSense/librealsense/issues/8562#issuecomment-813528142  that will help us in understanding your issue and provide support.

Thanks

bluebass44, 2021-04-13:
> Also if you could answer these questions #8562 (comment)

depth_sensor.set_option(RS2_OPTION_CONFIDENCE_THRESHOLD, 0);
 Negative impact - Are they less accurate?

RealSenseSupport, 2021-05-15:
Ah. I understand your question now. Yes. By allowing depth values to be calculated under lower confidence the tradeoff is  potentially less accurate values. The key is finding the right balance for your specific usage model.

Thanks

MartyG-RealSense, 2021-07-02:
Hi @bluebass44  Do you require further assistance with this case, please?  The case will be closed after 7 days from the time of writing this if we do not hear from you.  Thanks!

closed 2021-07-08 by MartyG-RealSense