realsenseai/librealsense #8563: Best camera for close range accuracy?

Darustc4, 2021-03-12:
Hello, currently I use a D435i camera for close range applications (<2 meters). The problem is that the depth point cloud is not precise enough; when pointing at a person they get a squashed nose, inaccurate proportions and other problems.

I would like to know if this camera is not suitable for my use case, or if it is me not configuring it correctly/processing the data incorrectly. 
In case the camera is simply not meant to be that precise at close range, which camera should I buy? I need HD color, depth and IMU data.

MartyG-RealSense, 2021-03-13:
Hi @Darustc4   The D435i has a minimum depth sensing range of 0.1 meters, compared to 0.3 m for D415 and 0.4 m for D455.  It is therefore an appropriate choice of camera model for close-range depth measurement (plus the D415 does not have an IMU).

If the camera is more than 0.4 m from the observed person then a D455 would be a viable option, as it contains an IMU.  For your stated use case though, your existing D435i can likely meet your project needs.

If you are observing the person with your own self-created program instead of the **RealSense Viewer** tool, then - unlike the Viewer - post-processing filters are not applied by default and must be deliberately programmed into the application.  So if you have written your own script, applying post-processing filters may provide an enhancement in image quality and accuracy.

As you are using color data, are you performing depth to color alignment or color to depth alignment in your project, please?

Also, could you describe the other problems in your image aside from inaccurate proportions?

Darustc4, 2021-03-16:
Thanks for the recommendation, Marty. I have checked in the Realsense Viewer and seen the root of the problem:
When using lower resolutions there are not enough points to properly define the depth map so a nose appears squashed, but when using full resolution this problem disappears.

Additionally, disabling the spatial filter helps details pop more, it seems like this filter is way to aggressive for high precision applications.

closed 2021-03-16 by Darustc4

MartyG-RealSense, 2021-03-16:
Thanks very much for the update @Darustc4 

A RealSense team member has previously agreed with your conclusions about the Spatial filter's processing demands.

https://github.com/IntelRealSense/librealsense/issues/4468#issuecomment-513485662