realsenseai/librealsense #8569: Raspbian installation instructions error, opencv suggested deb package fails to install

alonblade, 2021-03-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | All (SDK problem) | 
| Firmware Version                   | Irrelevant | 
| Operating System & Version | Raspbian  | 
| Kernel Version (Linux Only)    | 5.10.17-v7l+  | 
| Platform                                 | Raspberry Pi 4 |
| SDK Version                            | master (git describe says: v2.5.2-8567-g8fc9ad20d) | 
| Language                            |  Irrelevant | 
| Segment			| Irrelevant | 

### Issue Description
When following the instructions on https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md at the "Install opencv" section the instructions are to install https://github.com/mt08xx/files/raw/master/opencv-rpi/libopencv3_3.4.3-20180907.1_armhf.deb but this results in missing dependencies since the dependencies are no longer available in current (2020-03-14) raspbian:

```
root@blackian:/home/robot# dpkg -i libopencv3_3.4.3-20180907.1_armhf.deb
Selecting previously unselected package libopencv3.
(Reading database ... 110327 files and directories currently installed.)
Preparing to unpack libopencv3_3.4.3-20180907.1_armhf.deb ...
Unpacking libopencv3 (3.4.3-20180907.1) ...
dpkg: dependency problems prevent configuration of libopencv3:
 libopencv3 depends on libjasper1; however:
  Package libjasper1 is not installed.
 libopencv3 depends on libavcodec57; however:
  Package libavcodec57 is not installed.
 libopencv3 depends on libavformat57; however:
  Package libavformat57 is not installed.
 libopencv3 depends on libswscale4; however:
  Package libswscale4 is not installed.
 libopencv3 depends on libx264-148; however:
  Package libx264-148 is not installed.
 libopencv3 depends on libatlas3-base; however:
  Package libatlas3-base is not installed.
 libopencv3 depends on libcanberra-gtk0; however:
  Package libcanberra-gtk0 is not installed.
 libopencv3 depends on libavresample3; however:
  Package libavresample3 is not installed.
 libopencv3 depends on libtesseract3; however:
  Package libtesseract3 is not installed.
 libopencv3 depends on liblept5; however:
  Package liblept5 is not installed.
 libopencv3 depends on libhdf5-100; however:
  Package libhdf5-100 is not installed.

dpkg: error processing package libopencv3 (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 libopencv3
```

The current version of libopencv3 on raspbian is 3.2.0 - is that sufficient?

Thanks,

Alon

MartyG-RealSense, 2021-03-14:
Hi @alonblade   The documentation for the librealsense OpenCV wrapper states that its examples were designed and tested with OpenCV 3.4, so I would recommend using that as a minimum version.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv

This Raspbian guide was written with Raspberry Pi 3 in mind.  A Pi 4 installation guide that is popular with RealSense users is the **Acrobotic** guide.  Its structure is similar to the official guide for Pi 3, but it does not include the OpenCV section.

https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

Acrobotic instead provide a separate documentation page for OpenCV on Pi.  The **pip install opencv-contrib-python==4.1.0.25** installs OpenCV 4.1.  As the OpenCV wrapper documentation states that the OpenCV examples may need some editing to work with Python 4, you could try substituting that 4.1 version number for one of the 3.4 versions from the pip OpenCV archive

https://github.com/acrobotic/Ai_Demos_RPi/wiki/Gettings-Started-with-OpenCV

https://pypi.org/project/opencv-contrib-python/3.2.0.8/#history

It is worth mentioning that some RealSense users have found that installing a more recent version of **protobuf** than suggested by the Intel and Acrobotic guides, such as **3.14**, has worked well for them.

https://github.com/IntelRealSense/librealsense/issues/7927#issuecomment-740114741

alonblade, 2021-03-14:
Thanks for the fast reply!

If I understand correctly only the examples rely on OpenCV, not the core librealsense2.so?

On an unrelated note, I've built with '-j4' and did not see any issues with out of memory (the 2 G swap file was required, but never went over 1G usage when I examined it).

MartyG-RealSense, 2021-03-14:
The OpenCV examples require the building of support for them with a CMake flag called **BUILD_CV_EXAMPLES**.  If you do not want to use OpenCV though then the minimal librealsense build should work fine.

MartyG-RealSense, 2021-03-21:
Hi @alonblade   Do you require further assistance with this case, please?  Thanks!

alonblade, 2021-03-21:
no, we can close this.

MartyG-RealSense, 2021-03-21:
Thanks for the update!

closed 2021-03-21 by MartyG-RealSense