realsenseai/librealsense #8577: What is the unit of focal lenght from intrinsic parameters?

Oldog, 2021-03-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (05.12.09.00） | 
| Operating System & Version |   {Ubuntu 18.04} | 
| Kernel Version   |  (5.4.0-58-generic) |                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.40 }                          | 
| Language                            |  {python}                          | 
| Segment			|  {VR }                          | 

### Issue Description
I'd like to convert the depth map to dispairyt map in pixel unit (not in meter). So first I get the intrinsic parameters as following:
```
depth_map=frame.get_depth_frame()
depth_prof=depth_map.get_profile().as_video_stream_profile()
depth_intr=depth_prof.get_intrinsics()
```
Here is the result:
```
[ 1280x720  p[627.562 374.707]  f[636.763 636.356]  Inverse Brown Conrady [-0.0573339 0.0663474 0.000395056 -1.37368e-05 -0.0211517] ]
```
What's the unit of focal lenght f? And how can I get the size of CCD sensor size (then I can convert them into pixel unit)?

MartyG-RealSense, 2021-03-15:
Hi @Oldog   In terms of intrinsics, focal length is described by the **fx** and **fy** fields, whose unit is in pixels (pixel width and pixel height).

https://dev.intelrealsense.com/docs/projection-in-intel-realsense-sdk-20#section-intrinsic-camera-parameters

The focla length formula is also referenced in Intel's **Projection, Texture-Mapping and Occlusion** white-paper document.

https://dev.intelrealsense.com/docs/projection-texture-mapping-and-occlusion-with-intel-realsense-depth-cameras#section-2-2-projection-to-camera-coordinates

You can find the focal lengths in the **mm** unit for different sensors / camera models in the data sheet document for the 400 Series cameras.  If you do a 'find' operation in the document for the term **focal length** then you can easily browse through all the references to sensor focal length.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

Oldog, 2021-03-16:
> Hi @Oldog In terms of intrinsics, focal length is described by the **fx** and **fy** fields, whose unit is in pixels (pixel width and pixel height). [...]

I get it. Thanks you so much.

closed 2021-03-16 by Oldog