realsenseai/librealsense #8578: 3D Point transform from depth coordinate system to color coordinate system

AndreKest, 2021-03-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515  | 
| Operating System & Version |   Win 10   | 
| Platform                                 | PC  |
| SDK Version                            |  2.0                         | 
| Language                            |  python                          | 
| Segment			|  Robot                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hello everybody,

I got the intrinsics for the depth and the color camera 
I created a point cloud with help of the RealSense SDK functions:

rs_PC = rs.pointcloud()
rs_PC.map_to(color_frame)
rs_PointCloud = rs_PC.caluclate(depth_frame)
rs_PointCloud.export_to_ply(path, color_frame)

After this I work with PCL and Open3D to get center points of a plane. And then I want to transform this point to the color camera coordinate system but I point don't know in which coordinate system lies the point cloud. (from the depth sensor or the color sensor)

To get some background to the project:
With the help of the RGB camera and ArUco Marker I estimate the pose of a robot. This works fine so far. When I measure a point in the ArUco marker coordinate system I can send this point through the transformations to the color camera and then to the gripper coordinate system with the hand-eye calibration.  This works.

But I want to use the Lidar sensor from the L515 for depth information so I calculate a point cloud and get the center of a plane in this Pointcloud. This Works too. 

But now I want to send this point to the gripper, so I need this point in the color camera coordinate system. And there is the problem. 
Can someone tell me how I can transform the point? 
In which coordinate system lies the point cloud? The depth camera or the color camera coordinate system?

Thanks for your help.

AndreKest, 2021-03-16:
Case closed!
I had to rotate the saved .ply point cloud file with open3d around the x axes at 180° with the center point (0,0,0)

closed 2021-03-16 by AndreKest

zhangyang-intel, 2023-08-21:
Hi, i have this same case. But i don't know the reason for rotating the saved .ply point cloud file with open3d around the x axes at 180° with the center point (0,0,0).
Thanks for your reply!