realsenseai/librealsense #8579: GlobalTimeEnabled does nothing for the L500 series

bminortx, 2021-03-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L500 series | 
| Firmware Version                   | n/a | 
| Operating System & Version |  All  | 
| Kernel Version (Linux Only)    | n/a                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.42.0                          | 
| Language                            |  C                       | 
| Segment			|  Robot               | 

### Issue Description

The L500 series of devices is programatically set to not respond to the flag `GlobalTimeEnabled`. It seems this was a purposeful change, according to the commit 47ebc47f8. 

1. Is there documentation anywhere explaining this decision? 
2. There is currently no error or warning passed to the user if they do attempt to set this flag for the L500 series; it will just get ignored. How is this meant to be understood by the user without looking at the source?

RealSenseSupport, 2021-03-23:
Hi @bminortx 

There have been some timestamp changes regarding L515 camera.  One good place to look at latest and updates is in the release notes (https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2430) of the builds.  Also have a look at this PR, it may be of help:  https://github.com/IntelRealSense/librealsense/pull/7514

closed 2021-04-08 by RealSenseSupport

ThatGeoGuy, 2021-04-20:
Hey @RealSenseSupport, just wanted to ask that we re-open this. 

The issue at hand here is that the underlying library is silently ignoring a user option. Usually you might expect from the API:

1. If this option is invalid usually `rs2_supports_option` will return `false`. This does not seem to be the case.
2. The actual `rs2_set_option` call should return an error. This does not seem to be the case.

The issue here is legitimate: it is a silent failure of the API. In this case there's nothing documented to explain _why_ this occurs. We only found this out based on a report to our Rust wrapper, and then by digging into the source code itself. Anyways, there is nothing in the linked release notes outside of this mention:

>  #5202 - Disable GLOBAL_TIME option (Disable global timestamp by default for specific SKU) contributed by @aangerma 

Which hardly helps describe what is going on here unless you've read through the source yourself and looked specifically for this issue.