realsenseai/librealsense #8581: NodeJS support for L515

Joshpolansky, 2021-03-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 1.5.4.1 | 
| Operating System & Version |   Win 10 / MacOS  | 
| Platform                                 | PC  |
| Language                            |  nodejs                          | 
| Segment			|  Robot                         | 

### Issue Description

I am attempting to use node_librealsense to connect to an L515. Most the documentation for the L515 says it is supported in Node but I don't see the L515 mentioned anywhere in the Node documentation. I have a D435 that works with my application but an L515 and T265. 

I've tried in MacOS and Windows 10 (in a VM..)

Are they currently supported?

RealSenseSupport, 2021-03-26:
Hi @Joshpolansky 

We are looking into this and will provide an update soon.

RealSenseSupport, 2021-04-02:
Hi @Joshpolansky 

The latest release that just went out holds some updates for NodeJS.  Here is a link to the Release Notes for the 2.44 release:  https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2440

RealSenseSupport, 2021-04-08:
Hi @Joshpolansky 

Were you successful with the latest release?

If we don’t hear from you in 7 days, this issue will be closed.

Joshpolansky, 2021-04-08:
Hi @RealSenseSupport 

I haven't been able to test it yet. It looks like it's still in pre-release and I was expecting to see a NPM release for the update, since I haven't gotten the whole stack running to build it from source. Am I missing something?

RealSenseSupport, 2021-04-14:
You're right it is marked as a pre-release version but will see if we are missing anything.

Joshpolansky, 2021-04-27:
Hello! I finally got everything to build and it looks like everything is working now. Thanks!

closed 2021-04-27 by Joshpolansky