realsenseai/librealsense #8587: rs_imu_calibration failure

victorll998, 2021-03-16:
### Issue Description

Hi everyone, I got one new D455 and run it on a virtual machine that supports USB 3.0. Datastream works fine on both realsense-ros and realsense-viewer until I attempt to run the IMU calibration tool. 

It gives me an error like this when running the tool:
```
waiting for realsense device...

Error: No RealSense device.
Traceback (most recent call last):
  File "rs-imu-calibration.py", line 757, in <module>
    main()
  File "rs-imu-calibration.py", line 746, in main
    if os.name == 'posix' and old_settings is not None:
UnboundLocalError: local variable 'old_settings' referenced before assignmen
```
I followed the instruction and installed the python wrapper correctly. ```rs-emuerate-device``` can detect my camera. I wasn't sure it is related to the virtual machine environment or else. It would be helpful if you have any ideas related to this failure.  

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D455  | 
| Firmware Version                   | 05.12.11.00 | 
| Operating System & Version |    Linux (18)   | 
| Kernel Version (Linux Only)    |4.15.0-137-generic| 
| Platform                                 | PC  |
| SDK Version                            |   v 2.42.0                          | 
| Language                            |  C#                       | 
| Segment			|  Robot                     |

MartyG-RealSense, 2021-03-16:
Hi @victorll998   Did you test the Python wrapper with an example program after it was installed to confirm that it is working correctly, please?  If not, **python-tutorial-1-depth.py** would be a simple one to begin with.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-tutorial-1-depth.py

victorll998, 2021-03-16:
> Hi @victorll998 Did you test the Python wrapper with an example program after it was installed to confirm that it is working correctly, please? If not, **python-tutorial-1-depth.py** would be a [...]

I tested **python-tutorial-1-depth.py** and it is working fine

MartyG-RealSense, 2021-03-17:
I checked the documentation of the IMU calibration tool.  It states the requirement of **a USB type C cable to connect the device to the host computer**, implying that it expects a direct USB-cable connection to the computer that the tool is installed on instead of a VM connection.  

The documentation adds **Connect the device using the USB cable to the PC where Intel RealSense SDK 2.0 has been installed**.

victorll998, 2021-03-21:
Thank you @MartyG-RealSense. I have also noticed that. Probably the interface of VM machine causes this failure.

One thing I have found this week is that when I use USB 3.1 (type c to USB) instead of USB 3.2 (type c to type c) the problem is solved by itself. 

Does SDK2.0 fully support USB3.2?

MartyG-RealSense, 2021-03-21:
There was a past case where a USB 3.2 hub did not work with a 400 Series camera but a USB 3.1 hub had no problems.  That was a couple of years ago though and I have not heard of such issues since then.

closed 2021-04-07 by MartyG-RealSense