realsenseai/librealsense #8591: L515 - RGB Image undistortion

quizzy5889, 2021-03-16:
-------------------------------------------------------------------------------
| Required Info                 |                                                     |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.01.03 | 
| Operating System & Version | Ubuntu 18.04 LTS | 
| Kernel Version (Linux Only)    | 5.4.0-66 generic | 
| Platform                                 | PC |
| SDK Version                            | 2.40.0 | 
| Language                            |  opencv | 
| Segment			| Robot | 

### Issue Description
I am writing to you because I can not understand if the RGB camera image of my L515 is already rectified or not. I acquired some images of a calibrator and read through ROS the calibration parameters of the camera.

[image]

Then I used OpenCV in python to undistort the image:

```
camera_matrix = np.zeros(3, dtype=float)
camera_matrix[0, 0] = 904.15`
camera_matrix[0, 2] = 649.577
camera_matrix[1, 1] = 904.48
camera_matrix[1, 2] = 368.543
camera_matrix[2, 2] = 1

dist = np.zeros(shape=(1,5), dtype=float)
dist[0, 0] = 0.172075  #k1
dist[0, 1] = -0.549346 #k2 
dist[0, 2] = -0.0010658 #p1
dist[0, 3] = 1.46462e-05 #p2
dist[0, 4] = 0.477308 #k3

image_undistorted = cv2.undistort(image, camera_matrix, dist)
```

For some reason, I'm not sure if undistortion is applied correctly. For example, it seems to me that the left edge of the calibrator, after undistortion is slightly concave. The first image is the original, the second one is the undistorted one.

[image]

[image]

Could you please tell me if I'm making a mistake?

RealSenseSupport, 2021-03-26:
Hi @quizzy5889 

You can verify the distortion model for the L515 using the rs-enumerate-devices -c tool within the librealsense SDK.

Also want to point out that there is a pull request discussing and make changes to the distortion model that is identified when doing the rs-enumerate-devices command to get the distortion model.  Do review and look over this PR:  https://github.com/IntelRealSense/librealsense/pull/8523

RealSenseSupport, 2021-04-26:
Hi @quizzy5889 

Were you able to check out the PR?

closed 2021-05-07 by RealSenseSupport