realsenseai/librealsense #8592: Projected pattern gradually disappears in IR image

CodePointer, 2021-03-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435, D435i | 
| Firmware Version                   | 05.12.09.00 | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |  -  | 
| Platform                                 | PC  |
| SDK Version                            |  2.40.0 }                          | 
| Language                            | opencv/python }                          | 
| Segment			|  others  | 

### Issue Description
Dear Realsense team,

I run into a really weird problem when using D435i. I'm using two realsense device, D435i and D435. I want to observe the projected dot from D435, so I turn off the emitter of D435i and capture left & right infrared camera from D435i, as shown below:

[image]

But after several seconds, the projected dots are slowly erased:

[image]

Then after 10-20 seconds, the dots will gradually come back again. Such behavior repeat every 30-40 seconds, and I don't know how to stop it.

Some operations I take for debugging:

1. I tried python wrapper instead of realsense viewer, but it's the same.

2. I turned on D435i's emitter and then physically covered the emitter, still not works.

3. Turning off the color stream from D435i at the beginning can solve this problem. But I need RGB stream from D435i.

4. If I turn off the color stream after the dots vanished, the infrared camera won't observe projected dots again. Unless I disconnect D435i and operate like above.

Therefore, I believe there is some filtering or color balancing inside the D435i to remove noise from the captured image, but I cannot find the options in the realsense viewer. 

Can anyone help me with that? How could I get the infrared images with the pattern from D435's emitter?

MartyG-RealSense, 2021-03-16:
Hi @CodePointer   As a starting point in investigating your case, may I ask the reason for turning off the D435i's emitter please?  Multiple RealSense 400 Series cameras do not interfere with each other, so you should be able to have the IR emitters of both cameras active simultaneously without problems.

CodePointer, 2021-03-16:
Hi @MartyG-RealSense , thank you for your prompt reply!
I'm doing some research on the structured light system, where the emitter parameters are also important for me. I calibrated the parameters by myself and used the D435i's emitter at the beginning. But the baseline is too small for my researching. Thus I use another device to provide a longer baseline.
Therefore, I have to use the D435's emitter, and D435i's emitter has to be turned off. Although the stereo system can still work well, projected patterns interference will cause a lot of trouble to my research...

MartyG-RealSense, 2021-03-16:
Are you using the other device with the longer baseline in the same scene as the D435 and D435i cameras?  Although 400 Series cameras do not interfere with each other, they can interfere with non-RealSense devices.

CodePointer, 2021-03-16:
> Are you using the other device with the longer baseline in the same scene as the D435 and D435i cameras? Although 400 Series cameras do not interfere with each other, they can interfere with non-RealSense devices.

Nope, I only use these two devices for data collection.

[image]

MartyG-RealSense, 2021-03-16:
Are the lighting conditions in the room consistent or do they fluctuate?  I once handled a case where the visibility of the dots varied depending on how dim the lighting conditions were.  So in morning they might be visible on cloth but not wood, and later in the day the situation would reverse and they would be visible on wood but not cloth.  So if the lighting level was changing more rapidly than that then I could visualize how the dots could come and go.

CodePointer, 2021-03-16:
> Are the lighting conditions in the room consistent or do they fluctuate? I once handled a case where the visibility of the dots varied depending on how dim the lighting conditions were and the type of material the dots were projected onto. So in morning they might be visible on cloth but not wood, and later in the day the situation would reverse and they would be visible on wood but not cloth. So if the lighting level was changing more rapidly than that then I could visualize how the dots could come and go.

The lighting conditions are consistent. Besides, the D435i and D435 are placed on the tripod, which means the position is consistent too.

MartyG-RealSense, 2021-03-16:
Under the **Controls** section of the Viewer's Stereo Module side-panel options, could you try enabling the **Emitter Always On** option if you have not done so already, please?

[image]

CodePointer, 2021-03-16:
Hi @MartyG-RealSense , sorry for the late reply.

I tried your suggestion, but the problem still exists. 

Here is the snapshot before opening the color stream:

[image]

And here is the result several seconds (maybe 20-30s?) **after** enabling the color stream:

[image]

Besides, I also set the **Emitter Enabled** tag to **Laser**, and physically cover the emitter, the problem still exists...

CodePointer, 2021-03-16:
Hey @MartyG-RealSense ! 
Seems the problem solved. 

I set the **Emitter Always On** option of **D435**, and the projected dots can be observed correctly.

[image]

Now I'm finding explanations of this option in the SDK doc for this problem~

Thank you for your help!

MartyG-RealSense, 2021-03-16:
Great news that setting **Emitter Always On** worked for you - thanks very much for the update!

CodePointer, 2021-03-16:
> Great news that setting **Emitter Always On** worked for you - thanks very much for the update!

Thank you! But here is another small problem: I can't find **Emitter Always On** option in [Python Wrapper](https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.option.html?highlight=emitter#pyrealsense2.option.emitter_enabled). Do you know how to set this mode in python codes?

MartyG-RealSense, 2021-03-16:
My research indicates that the instruction for enabling this mode in Python is:

**sensor.set_option(rs.option.emitter_always_on, 1)**

CodePointer, 2021-03-16:
Thank you! Seems the python wrapper has this option while the documents miss this one.

Besides, this code works for me:

```
# - Package Imports - #
import pyrealsense2 as rs

# - Coding Part - #
device_sn = {'D435': 'xxxxxxxxxxxx', 'D435i': 'xxxxxxxxxxxx'}

configs = {'D435i': rs.config(), 'D435': rs.config()}
configs['D435i'].enable_device(device_sn['D435i'])
configs['D435i'].enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 15)
configs['D435i'].enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 15)
configs['D435i'].enable_stream(rs.stream.infrared, 1, 848, 480, rs.format.y8, 15)
configs['D435i'].enable_stream(rs.stream.infrared, 2, 848, 480, rs.format.y8, 15)
configs['D435'].enable_device(device_sn['D435'])
configs['D435'].enable_stream(rs.stream.infrared, 1, 424, 240, rs.format.y8, 15)

pipelines = {'D435i': rs.pipeline(), 'D435': rs.pipeline()}
profile_d435 = pipelines['D435'].start(configs['D435'])
profile_d435i = pipelines['D435i'].start(configs['D435i'])

try:
    device_435i = profile_d435i.get_device()
    depth_sensor = device_435i.query_sensors()[0]
    depth_sensor.set_option(rs.option.emitter_enabled, 0)

    device_435 = profile_d435.get_device()
    depth_sensor = device_435.query_sensors()[0]
    depth_sensor.set_option(rs.option.laser_power, 360)
    depth_sensor.set_option(rs.option.emitter_always_on, 1)

    while True:
        # Collect data here.
        pass

finally:
    pipelines['D435'].stop()
    pipelines['D435i'].stop()
```

Anyway, the problem solved! Thank you for your help~

closed 2021-03-16 by CodePointer

MartyG-RealSense, 2021-03-16:
You are welcome - thanks very much for sharing your solution with the RealSense community   :)