realsenseai/librealsense #8595: compatibility with VOXL autonomy board

nikhilarane, 2021-03-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Operating System & Version |   Linux (Ubuntu 18)  | 
| Platform                                 | Snapdragon 821  |

### Issue Description
We want to use RealSense D455 on ModalAI VOXL computer. They said that this will require a kernel patch. Is this possible with D455? Will RealSense supply that kernel patch?

Thanks

MartyG-RealSense, 2021-03-16:
Hi @nikhilarane  It souds as though you wish to connect the D455 camera hardware to the **ModalAI VOXL** and use it 'plug and play' (works immediately upon insertion) without the need to install additional software for the camera.  Is that correct, please?

A ModelAI team member stated on their forums in 2020 in response to a question about the RealSense **D435i** camera model that they did not 'currently' support RealSense at the time that the reply was written.

https://forum.modalai.com/topic/34/d345i-camera-support

So in a plug-and-play scenario where no additional software for the camera such as the **RealSense SDK (librealsense)** is being installed then it seems as though a RealSense camera such as a D435i or D455 will not function with this board.

There have been cases where a RealSense camera and PX4 has been used with an **Odroid** board instead.  Typically the RealSense model involved is the T265 Tracking Camera, though the link below at the **PX4** GitHub does have a tutorial project for setting up with a D435, Odroid board and Px4.

https://github.com/PX4/PX4-Avoidance 

If you do need to use a ModalAI VOXL with the D455 and would be using the camera 'plug and play' with no RealSense SDK software installed then I can ask Intel about patching if you wish.

nikhilarane, 2021-03-16:
Thanks for your reply. I'm ok with having to install RealSense SDK in order to get D455 to work on it. I am a little confused when the ModalAI engineer said that it would require a kernel patch. Not really sure what that means. If it means just installing the SDK, then it is hardly any work.

MartyG-RealSense, 2021-03-16:
If it was a matter of RealSense cameras being incompatible with the kernel that is used by ModalAI VOXL then the RealSense SDK provides a way around that problem.  The SDK can be compiled with an installation method called **RSUSB backend** that does not have dependence on Linux versions and kernel versions and does not require patching.  

The problem is that as there is no reference available regarding using RealSense cameras with ModalAI VOXL, it is difficult for me to say what impact this would have regarding whether the software installed on ModalAI VOXL would be able to interface with the camera.

nikhilarane, 2021-03-16:
Thanks for your help and suggestion. I will close this issue for now. Will get back if I face trouble trying your suggestion out.

closed 2021-03-16 by nikhilarane

MartyG-RealSense, 2021-03-16:
You are very welcome @nikhilarane - please do feel free to ask further questions if you need to.  Good luck!