realsenseai/librealsense #8597: "unmet dependencies" Ubuntu installation

zanellar, 2021-03-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 |  
| Operating System & Version |   Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  4.15.0-136-generic                                        | 
| Platform                                 | PC |
| SDK Version                            |  SDK 2.0 (v2.42.0)                         | 
| Language                            | python                         |  

### Issue Description

Hi, 

I am following instructions as on the following link : 
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md 

but i get this error 
> The following packages have unmet dependencies:
>  librealsense2-utils : Depends: libgtk-3-dev but it is not going to be installed

when I try to install `librealsense2-utils`

I've tried (as suggested in other threads)
```
sudo apt-get update
sudo apt-get upgrade
sudo apt-get -f install
sudo apt-get autoclean
```
Do you have any suggestion?

Thanks

MartyG-RealSense, 2021-03-16:
Hi @zanellar  Are you able to build successfully if you remove all librealsense packages and then repeat the build process, please?

The instruction for removing all librealsense packages is:

dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#uninstalling-the-packages

Can you also confirm please that when you install the packages, you use the Ubuntu 16 version of the instruction for adding the server to the list of repositories:

**sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main" -u**

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages

zanellar, 2021-03-17:
> Are you able to build successfully if you remove all librealsense SDK related packages and then repeat the installation process, please?

I've tried, but it still have the same problem (_librealsense2-utils : Depends: libgtk-3-dev but it is not going to be installed_)

> Can you also confirm please that when you install the packages, you use the Ubuntu 16 version of the instruction for adding the server to the list of repositories:
> 
> `sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main" -u`
> 

Yes, I've used that line.

MartyG-RealSense, 2021-03-17:
The link below suggests that this problem with libgtk-3 may be caused by the packages not being able to be found, rather than them having unmet dependencies:

https://askubuntu.com/questions/741988/unmet-dependencies-while-installing-gtk-3/742018#742018

zanellar, 2021-03-17:
I've already seen that answer, but how can I check if there is _"an active repository that is suggesting a newer version of libgtk-3-dev"_?

MartyG-RealSense, 2021-03-17:
The link below discusses meeting the dependency requirements with downgrades or upgrades.

https://stackoverflow.com/questions/38752214/installing-libgtk-3-dev-unmet-dependencies

In honesty, the subject of meeting libgtk-3-dev dependencies is so complicated though that it may be easier just to start with a fresh installation of Ubuntu in the hope that it clears the problem.

zanellar, 2021-03-18:
ok, i really think i will give up. Upgrading/downgrading packages gets quite tedious, there is a long chain of nested dependencies that seems endless. Moreover, I would probably mess some.

closed 2021-03-18 by zanellar

MartyG-RealSense, 2021-03-18:
Okay, thanks very much for the update.  Sorry to hear about your difficulties with ibgtk-3-dev