realsenseai/librealsense #8598: Q: Eye-Tracking Applications

lorenzleutgeb, 2021-03-16:
Hi!

I am considering buying an Intel RealSense D435 with the goal of using it as (a) 3D scanner, (b) head tracker, (c) eye tracker. I already own a Tobii 4C eye tracker, but Tobii's support for Linux, or in general more open use of their hardware, is very bad so here I am.

This issue is purely about eye tracking. I found [GazeSense](https://eyeware.tech/gazesense/) (commercial product) that seems to do eye tracking via Intel RealSense D435, claiming 2-5 degrees of accuracy.

Did anyone else implement eye tracking with an accuracy of <3 degrees with Intel RealSense (also with other devices, not just the D435)? Please point me at some code!

Somewhat related:
 - #7730 (mentions eye tracking code sample for SR300)
 - #8095 (talks about face tracking, which is related to eye tracking and head tracking more generally)

MartyG-RealSense, 2021-03-16:
Hi @lorenzleutgeb   **Eyeware GazeSense** is the main RealSense-compatible commercial solution available for eye / gaze tracking.  It has also been highlighted in official Intel RealSense articles.

https://www.intelrealsense.com/types-of-tracking-overview/

https://www.intelrealsense.com/software-for-intel-realsense/#eye-tracking

In regard to open-source solutions: if you are able to use Python then the **PyGaze** toolbox could provide a pre-made means of achieving eye tracking.

http://www.pygaze.org/

You could also use OpenCV and the **dlib** face recognition library, like the tutorial in the link below.

https://towardsdatascience.com/real-time-eye-tracking-using-opencv-and-dlib-b504ca724ac6

I recommend googling for **opencv eye gaze** to research other ways of implementing eye tracking with OpenCV.

lorenzleutgeb, 2021-03-16:
OK, thanks for those pointers! I also found [this similar discussion in Intel Community](https://community.intel.com/t5/Items-with-no-label/Gaze-tracking-with-the-Intel-realsense/td-p/520045) which also mentions GazeSense and the idea of integrating with PyGaze.

To be honest, I was hoping that there are more people experimenting with eye tracking on Intel RealSense, but that's okay :)

MartyG-RealSense, 2021-03-24:
Hi @lorenzleutgeb  Do you require further assistance with this case, please?   Thanks!

lorenzleutgeb, 2021-03-24:
No further assistance is required. I did not close the issue, hoping that other people might bring up more examples/applications/codebases, but that probably won't happen.

closed 2021-03-24 by lorenzleutgeb