realsenseai/librealsense #8601: Segmentation fault in reading frames of *.bag file

kir-ark, 2021-03-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.10.03.00 | 
| Operating System & Version |  Linux (Ubuntu 16) | 
| Kernel Version (Linux Only)    |  4.15.0-50-generic  | 
| Platform                                 | PC |
| SDK Version                            |  2.42.0.2924                          | 
| Language                            |  python                          | 
| Segment			|  others                          | 

This issue is similar to https://github.com/IntelRealSense/librealsense/issues/5503.
I extract frames from a *.bag file by index (index maps to time and passes to seek() function) and then display depth and color frames. Then at some point  when the pipeline stops (pipeline.stop()), my application crashes with **segmentation fault** error. I tried different bag files with different stream settings. Segmentation fault happens in a random time.

I attached the script that reproduces the problem. It plays a *.bag file frame by frame using seek() function to imitate access by index.
[bag_reader.py.zip](https://github.com/IntelRealSense/librealsense/files/6151461/bag_reader.py.zip)

MartyG-RealSense, 2021-03-17:
Hi @kir-ark   In past cases, this kind of error with bag frame scripts has tended to occur when reading bags using a loop.

It sounds as though you want to playback the bag file and display its contents on-screen, and not extract the frames into files.  If that is correct, the SDK's **read_bag_example.py** bag reading Python example program may meet your needs.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/read_bag_example.py

kir-ark, 2021-03-19:
I see your point, but that doesn't work for my purposes. This **read_bag_example.py** read frames one by one. Instead of that I need to read random frame at an arbitrary time. That is why I using seek() function. The example above **bag_reader.zip** is particular case just to show you the **segmetation fault** error. In more general case I need to read one random frame at a time. 

I suppose the crash happens because of using *registration* method.

Could you please explain is there any other method to read random frame at a time or figure out why **segmentation fault** occurs in using *registration* and *seek* function?

MartyG-RealSense, 2021-03-19:
Jumping to a specific frame in a bag during playback does not usually work well when **set_real_time = false**.  Starting at the first frame and skipping forward to the target frame has been recommended instead.  

https://github.com/IntelRealSense/librealsense/issues/3121#issuecomment-456476519

closed 2021-03-31 by MartyG-RealSense