realsenseai/librealsense #8603: Open the L515 RGB camera will be disconnected and then reconnected to the computer

kmolLin, 2021-03-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   |  01.05.04.01) | 
| Operating System & Version |   windows10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.42                     | 
| Language                            |  python & C }                          | 
| Segment			| other                          | 

### Issue Description

Hi I have problem with the L515 camera.

I can use the motion Module like the picture below.
[image]

but I can't open the RGB camera & Depth sensor camera.

I have checked these things
1. open the windows 10 camera privacy
[image]

2.  Add the real-sense SDK in system environment
[image]

3. Installation Manager show device successful

[image]

When I try to open the RGB camera, The camera will be disconnected and then reconnected to the computer ,
the RealSense Viewer console log show "types.h:315 - hr returned: HResult 0x800703e3: The I/O operation has been aborted because of either a thread exit or an application request", I have search same question but no one have solution.

[image]

[image]

[image]

kmolLin, 2021-03-17:
And I have try use python call the SDK start camera, 

import pyrealsense2 as rs
pipe = rs.pipeline()
profile = pipe.start()

the error is 

RuntimeError: MFCreateDeviceSource(_device_attrs, &_source) returned: HResult 0x80070003: "系統找不到指定的路徑。"

RealSenseSupport, 2021-03-23:
@kmolLin Do you have another L515 or RealSense camera to see if the same issue?

kmolLin, 2021-03-24:
I have tried D435I, and it work well.

RealSenseSupport, 2021-03-26:
@kmolLin Could you please try the latest SDK v2.43 and the latest FW v1.5.5.0 to see if you still have the same issue? Thanks!

https://github.com/IntelRealSense/librealsense/releases/tag/v2.43.0 
https://dev.intelrealsense.com/docs/firmware-releases-l500

kmolLin, 2021-03-29:
The product have been return to intel, I'm not sure it will be all right with new FW v1.5.5.0. I will close the issue. THX~

closed 2021-03-29 by kmolLin