realsenseai/librealsense #8607: Coral Dev Board Performance Problems

brunovollmer, 2021-03-17:
Required Info |  
-- | --
Camera Model | D435 + T265
Firmware Version | 05.12.09.00
Operating System & Version | Ubuntu 20.02
Kernel Version (Linux Only) | 5.4.0-56-generic
Platform | PC
SDK Version | legacy / 2.40.0
Language | Python
Segment | Robot

### Issue Description

Hey,

I'm currently trying to optimize the build process for librealsense on my Coral Dev Board. If I compare the performance my code on the Raspberry Pi 4 and the Coral Dev Board I see a difference of roughly 1.8-2 FPS. After some debugging I found the problem was related to the image and depth alignment process.

Currently I built the library with the following options on the Coral Dev Board:
`
cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DFORCE_RSUSB_BACKEND:bool=true -DBUILD_WITH_CUDA:bool=false -DBUILD_GRAPHICAL_EXAMPLES:bool=false -DCMAKE_BUILD_TYPE=release -DBUILD_WITH_OPENMP=true`

I think the alignment process is slower because it was mainly optimized for CUDA but does the Raspberry Pi support CUDA?

Any other suggestions on how to improve the built process to improve the performance or suggestions on how to skip the alignment? 

One idea I had was to just transfer the bounding boxes from the color image to the "full-size" depth image. This should computationally be cheaper. Any tips on how to accomplish that?

Thanks in advance,

Bruno

RealSenseSupport, 2021-03-23:
@brunovollmer What alignment do you mean? Which program did you test? And please note that when openmp enabled, it will reduce latency at expense of greater CPU utilization.

brunovollmer, 2021-03-23:
Hey @RealSenseCustomerSupport,

with alignment I mean the depth stream to color stream alignment. I'm running object detection on the color stream and need the exact depth values for each detected object thus I have to align the depth image to the color image.

RealSenseSupport, 2021-03-26:
@brunovollmer If you don't want do alignment, could you please try to run detection first and then use https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/rsutil.h#L186 to get the depth value for the detected object?

brunovollmer, 2021-03-29:
@RealSenseSupport Do you have documentation on what the function specifically does and what input it expects? Is it available in Python as well?

RealSenseSupport, 2021-03-31:
@brunovollmer Sorry we don't have such documentation. For python function please refer to https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/pyrsutil.cpp#L63 

Here's an example - https://github.com/IntelRealSense/librealsense/blob/master/common/post-processing-filter.h#L99

RealSenseSupport, 2021-04-06:
@brunovollmer Any other questions about this? Looking forward to your update. Thanks!

brunovollmer, 2021-04-13:
@RealSenseSupport I am unfortunately unable to further investigate in this manner. When I have time I will get back to you here.

RealSenseSupport, 2021-04-25:
@brunovollmer Any update from your side? Thanks!

RealSenseSupport, 2021-05-06:
@brunovollmer Did you get chance to try and any other questions? Looking forward to your update. Thanks!

RealSenseSupport, 2021-05-12:
@brunovollmer Any other questions? Please note that this ticket will be closed if we don't hear from you for another 7 days. Thanks!

brunovollmer, 2021-05-13:
No thanks

closed 2021-05-13 by brunovollmer