realsenseai/librealsense #8612: Unable to Detect Camera Ocassionally

ZainAli222, 2021-03-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | `05.11.06.250` | 
| Operating System & Version |  Linux (Ubuntu 18.04)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.9.140-tegra)                                         | 
| Platform                                 |  NVIDIA Jetson  |
| SDK Version                            |  2.36 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot }                          | 

### Issue Description

Querying the device connected to the device We are using 
```
vector<string> devs_sr;
rs2::context ctx = rs2::context();
rs2::device_list devices;
devices = ctx.query_devices();
```
It does detect the number of device connected but at time it is unable to detect one/two of the connected camera devices out of the four cameras. Restarting(Running it again) at times does detect them again. So can you guide what could be the cause or how it can be mitigated.

ZainAli222, 2021-03-22:
Thanks **MartyG-RealSense** for response, The problem is  `rs2Ctx.query_devices()` doesn't even detect one/two of the camera devices. We can reset the camera devices only if it detects them as camera device but in this case it doesn't sometimes detect a camera device or two. On restarting the application it does detect the device again at times but at times even restarting the application doesn't help

MartyG-RealSense, 2021-03-22:
If you are using Ubuntu, a way to reset the camera without it needing to be detected is to activate a Linux USB port reset script.  Would that be a viable option for your project, please?

closed 2021-04-12 by MartyG-RealSense