realsenseai/librealsense #8616: Best way to get RGBD?

Darustc4, 2021-03-18:
|---------------------------------|------------------------------------------- |
| Camera Model                        | D435i                                                      | 
| Firmware Version                    | 05.11.15.00                                             | 
| Operating System & Version  |   Raspbian Buster                                    | 
| Kernel Version (Linux Only)    |                                                                | 
| Platform                                  | Raspberry Pi                                           |
| SDK Version                            |  2.42.0                                                    | 
| Language                                |  {C++                                                     | 
| Segment			          |  AR                                                         | 

Hello, I am working on a real time volumetric data transmission over the internet program for AR.

Until now I have been creating a pointcloud and compressing it into a Draco3D file before transmitting it, but this is too resource intensive and slow (a Raspberry pi4 hardly can produce 20 fps at lowest resolution).
My next idea to try is to instead of sending a pointcloud where each vertex is XYZRGB (total of 120b), to send RGBD (32b). This is also very convenient because RGBD can be sent using RGBA video protocols, which offer very good compression.

I would like to know what is the best way to get this data without unneeded operations, for example, is it needed to map a pointcloud to the color frame and then calculate the points, which is shown in most examples? This seems unnecessary as each point is XYZ, while I just need the depth and color of each pixel in the image.

I am aware the **get_distance(int x, int y)** method exists, but I have read in https://dev.intelrealsense.com/docs/rs-measure that calling this function thousands of times per second will not be performant, so that means there has to be a better way.

Any examples or tips would be appreciated.

MartyG-RealSense, 2021-03-18:
Hi @Darustc4   There is a discussion in the link below that provides details of a range of resources for the kind of application that you have described.

https://github.com/IntelRealSense/librealsense/issues/6535

Alternatively, exporting a point cloud to a ply file with the instruction **save_to_ply** can provide a range of options for which elements of the cloud to include and exclude if you need to simplify it.

https://github.com/IntelRealSense/librealsense/issues/7564

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1save__to__ply.html

It is also worth mentioning that some RealSense users have dealt with the problem of performing processing operations such as align and post-processing on a low-end developer board by transmitting the raw camera data to the host computer receiving the data and then performing the processing on the more powerful host computer.

Darustc4, 2021-03-22:
Thanks for quick response. Looking into the resources you have linked I gather I can get the raw depth data using:

```c++
rs2::depth_frame dpt_frame = frame.as<rs2::depth_frame>();
const uint16_t *dpt_image = reinterpret_cast<const uint16_t*>(dpt_frame.get_data());
```
And also get the color in a very similar way but instead of unsigned 16 bit integers, you retrieve tuples of 3 unsigned 8 bit integers (RGB).

This is exactly what I needed but now a question appears: Is this depth distorted, or is it already ready to display? I am now skipping creating a pointcloud, mapping the color and calculating it's depth, and I am not sure if these steps correct some internal distortion of the camera or it is just a way to turn the more abstract 16b depth into <x,y,z> points. The code I no longer use is the following:
```c++
rs2::pointcloud pc;
pc.map_to(vid_frame);
rs2::points points = pc.calculate(dpt_frame);
```
If this code makes no important corrections to the depth, then I think everything is good to go.

MartyG-RealSense, 2021-03-22:
To the best of my knowledge, the **rs2::pointcloud pc** process is just generating a 3D point cloud from the depth frame.

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1pointcloud.html

The depth data would have undergone **rectification** inside the camera hardware at the point of capture before being sent along the USB cable to the computer / computing device.

closed 2021-03-22 by Darustc4