realsenseai/librealsense #8617: Error opening RealSense Viewer - Intel i7-3770K

frogzlegz, 2021-03-18:
| | |
|---|---|
|**librealsense**|2.42.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Illegal CPU instruction!
Possibly newer CPU architecture is required
```

MartyG-RealSense, 2021-03-18:
Hi @frogzlegz  The number '3' in your processor number 3770K indicates that it is a 3rd Generation Intel Core CPU.  From RealSense SDK version 2.36.0 onward, the version of the RealSense Viewer that is bundled with the SDK has required a 6th Generation (Skylake) processor or newer as it includes **OpenVINO**, which requires the AVX instruction set.

The standalone version of the RealSense Viewer does not have this requirement and can be downloaded from the **Assets** list at the bottom of the information listing for each SDK version on the SDK **Releases** page.  

You should launch the standalone Viewer from the location that you downloaded it to, or create a new desktop shortcut.  This is because the existing desktop shortcut for the Viewer will continue to point to the version with the CPU requirements.

https://github.com/IntelRealSense/librealsense/releases/

[image]

MartyG-RealSense, 2021-03-24:
Case closed due to correct answer provided and no further comments received.

closed 2021-03-24 by MartyG-RealSense