realsenseai/librealsense #8618: Realsense T265 not recognized with lsusb or with rs-enumerate devices

raghavauppuluri13, 2021-03-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | Newest Firmware (cannot get this info at the moment) | 
| Operating System & Version |  GNU/Linux  | 
| Kernel Version (Linux Only)    |  4.9.201-tegra                                        | 
| Platform                                 | Jetson Xavier NX Developer Kit |
| SDK Version                            |            2.28               | 
| Segment			|  Robot, ROS |                          | 

### Issue Description
On a fresh flash of Jetson Xavier NX developer kit, and following official install instructions using apt, I am unable to get the camera to even show up using `lsusb`. I have tried restarting and retrying multiple times without any success. I have also reflashed the system again without success as well. Any tips on what I should try next or where the issue might lie?

raghavauppuluri13, 2021-03-23:
Please any advice on this would help a lot.

RealSenseSupport, 2021-03-24:
Is the issue only happening on this particular platform with this setup?  Have you tried any other system like NUC or system with Windows or Ubuntu?  How about RSUSB backend, have you tried this?

raghavauppuluri13, 2021-03-24:
This occurred on an ubuntu 20.04 system, and a Jetson Nano platform. When referring to RSUSB backend, does that mean a build from source?

raghavauppuluri13, 2021-04-03:
Closing as it was a device issue. Received a new replacement that works as intended.

closed 2021-04-03 by raghavauppuluri13