realsenseai/librealsense #8620: L515 not being detected on realsense-viewer on Ubuntu 20.04

BhavyanshM, 2021-03-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (1.5.2.0 Last Known Version) | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  (5.8.0)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { v2.40.0 }                          | 
| Language                            |  {n/a}                          | 
| Segment			|  { n/a }                          | 

### Issue Description
The L515 is not being recognized as USB device or being detected by realsense-viewer or rs-enumerate-devices. It almost feels like it is a hardware failure. I have already looked through other similar Github issues. Please advice.

RealSenseSupport, 2021-03-23:
Hi @BhavyanshM 

Do you have any images that you can share?

Also are you using the USB cable that came with the camera?
Is it possible to use a Windows based machine with a USB3 port and see if the camera is recognized on that system?  Use also RealSense Viewer for Windows as well.
Have you tried flipping the orientation of the USB-C connector?

BhavyanshM, 2021-03-30:
[image]
[image]
[image]

Yes, using the MicroUSBC to USB 3.0 cable that came with the L515, the results were the same on Windows 10 as well. No USB device detected and RealSense Viewer did not show anything. The images were included above. Flipping the orientation of the USBC-C connector had no effect as well. Please advice. Thank you again.

closed 2021-04-09 by RealSenseSupport

pilleatus, 2022-01-18:
Flipping the orientation of the USB-C connector worked for me