realsenseai/librealsense #8621: D455 2.42.0 Driver does not work on Win7

mediadog, 2021-03-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455| 
| Firmware Version                   | Can't tell | 
| Operating System & Version |   Win 7 x64 Enterprise  | 
| Platform                                 | PC |
| SDK Version                            | 2.42.0 | 

### Issue Description
Just got a D455, and plugged it into a Win7 system that works fine with a D435.  It showed up under "Imaging Devices" in device manager, and I then updated the SDK to 2.42.0, making sure that the Win7 driver install option was checked.  I was prompted to OK the drier install, did so, and the device manager window updated a couple of times, but when done the D455 RGB and Depth were still listed under Imaging Devices, and the D455 is not recognized by anything.  The D435 still works fine.

I plugged the D45 into a Win10 system, updated to 2.42.0, and it works fine there.

mediadog, 2021-03-20:
Looked into this some more, and seems the driver installed with 2.42 is still 2.30 from 2019.  The .inf file for it does not have a listing for the D455, so it is not getting recognized.

MartyG-RealSense, 2021-03-20:
Hi @mediadog   The IMU-equipped **D435i** is listed in the SDK release notes as not supported by Windows 7 yet.  So whilst I have not seen previous reports of issues with D455 in Windows 7, this may be related to the D455 also being an IMU-equipped model.

[image]

MartyG-RealSense, 2021-03-31:
Hi @mediadog  Do you require further assistance with this case, please?   Thanks!

mediadog, 2021-03-31:
Well, is it fixed?  If the answer is "no", then my answer is "yes".

In fact, the D435i does appear in the Win7 driver .inf file.  But the D455 does not.  I hacked the .inf file to add the D455, and after doing a "have disk" update of the driver, the D455 did in fact get recognized and show up in device manager - but reaslsense-viewer does not recognize it.  So I suspect it is a Win7 librealsense issue as well as driver issue.

I was going to try hacking the .inf file so it would be reported as a D435, to see if the viewer would then recognize it, but have not had time.

Just curious, so when a bug is reported, is there any effort made to replicate it and schedule a fix?

MartyG-RealSense, 2021-03-31:
There is no information available regarding when D435i support for Windows 7 might be supported.  If the Release notes state that support of a particular RealSense camera model for a particular platform is scheduled for a future release then usually that support is implemented but no time schedule estimate can be provided.

Confirmed bugs are taken seriously, yes.  A rapid solution is not possible in every case though.

mediadog, 2021-03-31:
Well this issue is concerning the D455, not the D435i so your comment is perhaps in the wrong thread.  You mentioned the D435i support as perhaps the D455 being related because of the IMU.  I mentioned the D435i is in the driver .inf file just as an aside.  And when I modified the .inf file for the D455, the driver did install for it without errors, but the viewer still did not recognize it, so it may be more than a driver issue, perhaps the Win7 library version not recognizing the model number (or just not looking for it), but I am only guessing (it works fine on Win10, but there it appears to be using a generic Windows driver).

And in any case, if it is an IMU support issue, in the cases where the D455 could be used to enhance performance in D435 applications (like mine), transitional support for the image streams would be a huge win as the D455 is a great improvement over the D435.  I am now not going to buy any more D435s unless I absolutely have to, delaying any purchases pending D455 support (and I have some very large installations on the horizon).

Where is the roadmap for targeted release features and device support?  Thanks.

MartyG-RealSense, 2021-03-31:
Sorry, I meant D455.  This was in relation to my earlier comment in this discussion about how the reason for lack of D455 support on Windows 7 is likely for the same technical reasons that it is not supported on D435i when the D435 is.

https://github.com/IntelRealSense/librealsense/issues/8621#issuecomment-803265777

Intel do not publish roadmaps for camera features.  They are typically announced in the release notes at their time of introduction instead of schedule estimates being provided in advance.

mediadog, 2021-04-04:
I just looked at the v2.43.0 release page, and it still lists the D455 as supported, with the Win7 exception being just the D435i.  At least the release notes should show the D455 as not supported also.

MartyG-RealSense, 2021-04-04:
I have passed your suggestion about the release notes to Intel.  Thanks very much!

MartyG-RealSense, 2021-04-06:
Hi @mediadog   An official feature request has been filed with Intel for the addition of Windows 7 support of the D455.  This case should be kept open for whilst the request is being considered by the Intel RealSense team.  Thanks again!

noirflux, 2021-04-06:
Most excellent!  Athankew.

MartyG-RealSense, 2021-06-14:
Hi @mediadog  In regard to the feature request for D455 support on Windows 7, could you provide a reason why you are working with Windows 7 instead of Windows 10 please?  Intel's Windows 7 installation documentation makes the statement that "Not all product features are enabled on Windows 7".

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_win7.md

MartyG-RealSense, 2021-06-20:
Hi @mediadog   Could you provide information about the question in the comment above, please?  Thanks!

MartyG-RealSense, 2021-07-01:
Hi @mediadog  This case and the feature request will be closed if we do not receive a response to the earlier question in https://github.com/IntelRealSense/librealsense/issues/8621#issuecomment-860866222 after 7 days from the time of writing this.  Thanks very much.

mediadog, 2021-07-06:
So far trying to use Win10 for 24/7/.365 applications is problematic.  We have Win7 based installations using Kinect V2s with libfreenect2/OpenNI2 that have literally run constantly for over a year, only rebooting because of a power failure.  The amount of work required to try to get Win10 remotely stable enough for long term unattended use is considerable, and requires constant re-qualification.  We're talking public million-dollar-plus installations that can't have downtime because we missed plugging some Win10 "feature" or update.

MartyG-RealSense, 2021-07-06:
Thanks very much @mediadog for the explanation. I will pass it back to Intel.

MartyG-RealSense, 2022-08-24:
Closed due to ending of Windows 7 support from SDK 2.51.0 onwards.

closed 2022-08-24 by MartyG-RealSense