realsenseai/librealsense #8628: Announcement: RealSense SDK version 2.43.0 now available

MartyG-RealSense, 2021-03-21:
Hi everyone,

Version **2.43.0** of the RealSense SDK is now available from the link below.

Version https://github.com/IntelRealSense/librealsense/releases/tag/v2.43.0

**Release Notes**
https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2430

sam598, 2021-03-22:
Thanks for the update @MartyG-RealSense !

Multi-camera color sync is still not supported in any mode on any camera, correct?

MartyG-RealSense, 2021-03-22:
@sam598   It is not listed in the SDK release notes.  I have not seen the release notes for the new **5.12.12.100** firmware driver at the time of writing this.

jb455, 2021-03-22:
Is there any info available on the new thermal calibration please? The API Changes doc says "supported by selected Realsense Stereo devices" - which devices, and what is its purpose?

MartyG-RealSense, 2021-03-22:
Hi @jb455   The main SDK release notes for 2.43.0 state that it is for the D455 model.

"D455 introduces a compensation mechanism intended to mitigate the effect of thermal propagation in optics.  When active (default = On) it will track and adjust Depth and RGB calibration parameters automatically".

closed 2021-03-24 by MartyG-RealSense