realsenseai/librealsense #8630: Stream setting RS2_FORMAT_DISTANCE cannot be resolved

jimekevin, 2021-03-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.12.11.00 | 
| Operating System & Version |   Win 10  |                             | 
| Platform                                 | PC  |
| SDK Version                            | 2.40.0#1 (vcpkg installation)                          | 
| Language                            |  C++                          | 
| Segment			|  General | 

### Issue Description
I am currently porting an old project from the Kinect v2 to our new Realsense D415. The old code did a lot of calculations with the depth values in meters. So far, the code I am working with for the realsense camera is very very basic as I am trying to read into it and looking at the various examples. In the documentations, I found that both RS2_FORMAT_Z16 and RS2_FORMAT_DISTANCE return the distance in meters. The first seemed to work fine, but the second threw an unknown error. After some digging, I found that ? `cfg.can_resolve(pipe_)` returns false and the program crashes when calling `pipe.start()`. Why is that so? And what is the difference between the two? Ultimately I will convert the depth and rgb streams to OpenCV matrices.

```
	rs2::pipeline pipe;
	rs2::config cfg;
	cfg.enable_stream(RS2_STREAM_DEPTH, RS2_FORMAT_DISTANCE);
	cfg.enable_stream(RS2_STREAM_COLOR, RS2_FORMAT_RGB8);     // RS2_FORMAT_RGBA8
	cfg.can_resolve(pipe); // returns false
	pipe.start(cfg); // throws error without description
```

MartyG-RealSense, 2021-03-22:
Hi @jimekevin   There are only a few times that **RS2_FORMAT_DISTANCE** has been used since its introduction in the RealSense SDK, and never in a cfg definition.  I would therefore recommend using the Z16 format instead, which is the standard format for the depth stream in the SDK (no other format is usually used for the depth stream).

In regard to the difference between the instructions, Z16 is described as a **16-bit linear depth value** and DISTANCE as a **32-bit float-point depth distance value**.

A very rare example of use of RS2_FORMAT_DISTANCE is in the link below

https://github.com/IntelRealSense/librealsense/blob/master/src/proc/units-transform.cpp#L24

jimekevin, 2021-03-30:
Thanks for your response! Everything works fine with Z16 but I am still curious as to what needs to be done to be able to use **RS2_FORMAT_DISTANCE**. The documentation on it is very sparse. I mean, it must exist for a reason, right?

MartyG-RealSense, 2021-04-06:
Hi @jimekevin  Do you require further assistance with this case, please?  Thanks!

jimekevin, 2021-04-10:
Hi @MartyG-RealSense, yes my personal problem is resolved.

I was not able to get the other mode to work. I won't use it for now but I feel like there is still either something not documented with this or there are in fact  some bugs with this new mode. It doesn't matter too much for me right now because Z16 works just fine. Still, maybe someone else is also interested in this one and can elaborate on it. Thanks so far anyway!

closed 2021-04-10 by jimekevin