realsenseai/librealsense #8631: t265 RealSense error calling rs2_get_frame_width(frame_ref:nullptr):     null pointer passed for argument "frame_ref"

hongSS0919, 2021-03-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

RealSenseSupport, 2021-03-24:
Thanks for submitting this issue.  Are you only seeing this on the hello sample?  Or are you seeing this issue with any other samples?  Are you seeing this with the RealSense Viewer?

hongSS0919, 2021-03-25:
1. I saw the image and tracking track on the realsense viewer.I ran your other cases, ar-basic, callback, pose, capture, sensor-control, software-device, pose-predict, save-to-disk, all of which were successful .
2. Cases hello-realsense, align, align-advanced , Pointcloud these projects, the terminal will be closed after a flash, I call it "flash back" here, the program did not report an error (in some cases, I forgot to run the results).
3. Run hello-realsense on my own new project, it has the null pointer problem I mentioned above.

eszabo12, 2021-08-14:
Also getting this issue

closed 2026-10-23 by MartyG-RealSense