realsenseai/librealsense #8634: SLAM module not working in realsense D435(Unity).

AshwathamanGovind, 2021-03-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win (10) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  legacy / 2.0 (v2.42.0)                       | 
| Language                            |  unity 2019.4                      | 
| Segment			| VR (Oculus Rift) }                          | 

### Issue Description

 I am trying to run the SLAM module in unity .I am not able to get the streaming data from realsense D435 in SLAM module alone. Initially I had to copy paste the pre-built .dll from real sense SDK to get other 2d modules running. How to get those SLAM  module up and running? Any help from your side would be of great help. Thanks in advance.

MartyG-RealSense, 2021-03-22:
Hi @AshwathamanGovind   The SLAM example in the RealSense Unity wrapper is for the RealSense T265 Tracking Camera model and not the 400 Series stereo depth cameras such as D435.

It is important to note that the D435 model does not have an IMU component (whilst the D435i model does).  A popular SLAM solution that has been used with the D435 is **ORB_SLAM2**.

https://github.com/raulmur/ORB_SLAM2

There was a Unity implementation of ORB_SLAM2 created several years ago.  I do not know how well it would work in current Unity versions.

https://github.com/VisionerTech/ORB_SLAM2_Unity

MartyG-RealSense, 2021-03-30:
Hi @AshwathamanGovind  Do you require further assistance with this case, please?  Thanks!

AshwathamanGovind, 2021-03-30:
hai.. @MartyG-RealSense  The above mentioned SLAM App doesn't go well with the newer versions of unity. I am targeting unity 2019.4. Can you provide me some application that is compatible with that version of unity??

MartyG-RealSense, 2021-03-30:
i @AshwathamanGovind   I again researched your question thoroughly.  I could again find almost no modern option for using a RealSense 400 Series camera for SLAM in Unity though.

I expanded the search to be more general instead of specific to RealSense.  There is a commercial application called **Wikitude SLAM** that can be used with Windows and Unity.

https://www.wikitude.com/wikitude-slam/

The link below from Wikitude's blog has a example from February 2020 of use of the software in Unity.

https://www.wikitude.com/blog-developer-tutorial-slam-app-made-with-unity/

As Wikitude SLAM has a free trial version, it may be worth trying it with your D435 camera to see whether the software can access it.

AshwathamanGovind, 2021-04-02:
@MartyG-RealSense I tried the SLAM problem in ROS using the same RealSense D435, it is working fine. Is there a specific reason why it supports ROS, while it doesn't support the Unity platform irrespective of the lack of IMU component? Can you help me understand why is that the case?

MartyG-RealSense, 2021-04-02:
Reading data from cameras is something that is natural for ROS, so creating applications that do so is relatively straightforward.  Building camera applications in Unity is not so straightforward, as the usual approach to rendering camera data on-screen in Unity is by converting it into materials that can be applied onto object surfaces instead of enabling the raw values to be accessible.   For example, if you look at point cloud generator programs on the Unity Asset Store then they never show on-screen coordinates in their store screenshots.

AshwathamanGovind, 2021-04-04:
Thank you for your response.

Regards,
Aswath G

AshwathamanGovind, 2021-04-11:
No, I am not.
Thanks for your support. Really appreciate it.

Thanks & Regards,
Aswath G

MartyG-RealSense, 2021-04-11:
You are very welcome @AshwathamanGovind - thanks for the update!  

I will close the case.  Thanks again!

closed 2021-04-11 by MartyG-RealSense