realsenseai/librealsense #8635: I'm currently using d455 on the Jetson Nvidia Xavier(ubuntu). I'm facing incomplete frame detected and error with the motion sensor module kindly do check

prahalad12345, 2021-03-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2021-03-22:
Hi @prahalad12345   As a starting point in investigating your case, may I ask please if you are using the **RealSense Viewer** program?  And if you are using the Viewer, are you enabling the Depth and RGB streams in the RealSense Viewer at the same time as the Motion Module is enabled, or just the Motion Module on its own?  Thanks!

prahalad12345, 2021-03-22:
[image]

MartyG-RealSense, 2021-03-22:
The **Motion Module Failure** error is not a serious problem, just an annoyance as the Motion Module continues to work normally after the warning box is closed.

Having RGB, Depth and Motion Module all enabled simultaneously in the Viewer can cause errors for some RealSense users though.  It does not matter which order the three streams are enabled in, as activation of the Motion Module is the trigger for the problem.  There is no fix for this in the RealSense Viewer program, so if you do encounter problems then I would recommend enabling only two streams in the same session (e,g depth / RGB, depth / IMU, etc).

The **rs-multicam** example program, which works with a single camera too and can also display depth, RGB and Motion Module streams simulteneously (but does not have interactive controls) is not affected by this issue.

If you are programming oyur own application, a workaround for the depth-RGB-IMU issue is to put IMU and depth / RGB on separate pipelines (IMU alone on one pipeline and depth / RGB on the other pipeline) or by using **callbacks**.

MartyG-RealSense, 2021-03-28:
Hi @prahalad12345  Do you require further assistance with this case, please?  Thanks!

prahalad12345, 2021-03-28:
I'm all done thank you

MartyG-RealSense, 2021-03-28:
Okay, thanks very much for the update!   I will close this case.

closed 2021-03-28 by MartyG-RealSense