realsenseai/librealsense #8636: How to install it correctly static_node

vnadein, 2021-03-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200  } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Raspberry Pi  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I tried to do as in the documentation:

pose_sensor.set_static_node(nameNode, [data.translation], [data.rotation])

but get exception:

Invoked with: <pyrealsense2.pose_sensor object at 0xb366e2e0>, '1', [x: -0.000103101, y: 0.000634256, z: 0.000208607], [x: 0.00171485, y: 0.00043621, z: -0.00597328, w: 0.999981]

MartyG-RealSense, 2021-03-22:
Hi @vnadein   There are some potential problems with your project.

1.  RealSense SDK 2.0 does not support the R200 camera model.  So if you have an R200, the older **Legacy Librealsense** SDK should be used.

https://github.com/IntelRealSense/librealsense/tree/v1.12.1

2.  If you are using the Legacy version of librealsense, then using pyrealsense2 would not work either as it is designed for RealSense SDK 2.0.  For the Legacy SDK, the similarly named but different **pyrealsense** could be used.

https://github.com/toinsson/pyrealsense

3.  The biggest problem may be that the pose instructions are designed for the RealSense T265 Tracking Camera and are not supported by the R200 camera.

vnadein, 2021-03-22:
> Hi @vnadein There are some potential problems with your project.
> 
> 1. If you are using a RealSense R200 camera as indicated in the header of this case, RealSense SDK 2.0 does not support the R200 camera model.  So if you have an R200, the older **Legacy Librealsense** SDK should be used.
> 
> https://github.com/IntelRealSense/librealsense/tree/v1.12.1
> 
> 1. If you are using the Legacy version of librealsense, then using pyrealsense2 would not work either as it is designed for RealSense SDK 2.0.  For the Legacy SDK, the similarly named but different **pyrealsense** should be used.
> 
> https://github.com/toinsson/pyrealsense
> 
> 1. The biggest problem may be that the pose instructions are designed for the RealSense T265 Tracking Camera and are not supported by the R200 camera.
> 
> It is possible to obtain pose with a stereo depth camera using a **SolvePNP algorithm** in OpenCV or with **inference** in ROS.

Sorry, overshoot I made a mistake, i'm use T265

MartyG-RealSense, 2021-03-22:
Okay, thank you.  I am not involved in handling T265 cases.  So the best way to handle this will be to close this case and please ask you to post your opening message again as a new case on this forum, and including T265 in the message title.  My apologies for the inconvenience.

closed 2021-03-22 by MartyG-RealSense