realsenseai/librealsense #8638: Stream availability on Android devices

jb455, 2021-03-22:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Operating System & Version |   Android (various) | 
| Language                            |  Java                         | 

### Issue Description
Hi,

We have been testing our software on different Android devices and have noticed that not all stream configurations are available on all devices - eg, plugging the same camera into different Android phones/tablets, 1280*720*15 depth is available on some but not others. I've had a look through the code but I can't see how/where the SDK decides which stream profiles to return for a given device. Does it check USB bandwidth/memory availability/CPU speed somewhere and decide based on that? Is this overridable either within the standard SDK or by customising the SDK?
Thanks,
James

MartyG-RealSense, 2021-03-22:
Hi @jb455  I would recommend first checking whether the camera is being enumerated as USB 2.1 on a particular device.  This is because on a USB 2 connection, only **6 FPS** is available at 1280x720 resolution (as described in the link below regarding the 400 Series cameras' capabilities on USB 2).

https://www.intelrealsense.com/usb2-support-for-intel-realsense-technology/

Even if an Android device supports USB 3, the camera may sometimes be mis-detected as a USB 2 connection, as mentioned by Android wrapper developer **matkatz** in this discussion about USB 2 enumeration on USB 3-supported Android devices:

https://github.com/IntelRealSense/librealsense/issues/4091#issuecomment-500792067

MartyG-RealSense, 2021-04-12:
Hi @jb455  Do you have an update that you can provide about this case, please?

closed 2021-04-18 by MartyG-RealSense