realsenseai/librealsense #8641: I'm running the d455 camera using my Jetson Xavier(ubuntu 18) i tried running the terminal code for adding the servers to the list of repositories and I ended up with this

prahalad12345, 2021-03-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

prahalad12345, 2021-03-22:
Err:27 http://realsens-hw-public.s3.amazonaws.com/Debian/apt-repo xenial Release
  404  Not Found [IP: 52.216.134.75 80]
Hit:28 https://repos.codelite.org/ubuntu focal InRelease                       
Reading package lists... Done                       
N: Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'https://librealsense.intel.com/Debian/apt-repo focal InRelease' doesn't support architecture 'arm64'
E: The repository 'http://realsens-hw-public.s3.amazonaws.com/Debian/apt-repo xenial Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
N: Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo focal InRelease' doesn't support architecture 'arm64'
N: Skipping acquire of configured file 'universe/binary-arm64/Packages' as repository 'https://repos.codelite.org/ubuntu focal InRelease' doesn't support architecture 'arm64'

prahalad12345, 2021-03-22:
kindly do check

MartyG-RealSense, 2021-03-22:
Hi @prahalad12345   Are you attempting to install librealsense from Debian packages on a computer or computing device with an Arm processor please, such as a Raspberry Pi or Nvidia Jetson?  And are you doing so with the instructions in the link below?

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

If you are using an Arm device, the Debian packages from this repository will not work with Arm as they require an x86 or x64 architecture processor, such as those used in laptop and desktop computers.  Could you tell me which device you are using please (Pi, Jetson, etc) so that I can direct you to appropriate librealsense installation instructions for that device.

prahalad12345, 2021-03-23:
it an arm processor

MartyG-RealSense, 2021-03-23:
Thank you very much.  Which type of device is it please (Pi, Jetson, etc).

prahalad12345, 2021-03-23:
i've resolved it  using another post anyways thank you

MartyG-RealSense, 2021-03-23:
Excellent news that you found a solution - thanks for the update!

MrOCW, 2021-04-03:
> Hi @prahalad12345 Are you attempting to install librealsense from Debian packages on a computer or computing device with an Arm processor please, such as a Raspberry Pi or Nvidia Jetson? And are you [...]

HI @MartyG-RealSense, I am using both a Jetson Nano and Jetson Xavier NX, may I know the appropriate installation steps please?

MartyG-RealSense, 2026-10-23:
Case closed due to solution achieved and no further comments received after a long time period.

closed 2026-10-23 by MartyG-RealSense