realsenseai/librealsense #8642: Z-axis orientation in saved PLY file

AndreKest, 2021-03-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Operating System & Version |   Win 10  |                                      | 
| Platform                                 | PC  |
| SDK Version                            |  2.41 }                          | 
| Language                            |  python                         | 
| Segment			|  Robot                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Referring to this post: https://github.com/IntelRealSense/librealsense/issues/3589
There is standing that you changed the orientation from a right handed to a left handed coordinate system when I save the point cloud as PLY file.

But I need the right handed. How can I convert it back to a right handed coordinate system?

MartyG-RealSense, 2021-06-27:
Hi @AndreKest  Do you still require assistance with this case, please?  Thanks!

AndreKest, 2021-06-27:
@MartyG-RealSense No, I found a solution for my case. Thank you

MartyG-RealSense, 2021-06-27:
Thanks very much @AndreKest - that's great to hear!  I will close this case as you found a solution.

closed 2021-06-27 by MartyG-RealSense