realsenseai/librealsense #8643: Question: Do D435's color and depth have the same clock?

Egor18, 2021-03-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 

Hi!
I'm wondering if `rs::frame::get_timestamp()` for depth and color streams uses the same clock on D435?
(Assuming we're using `RS2_TIMESTAMP_DOMAIN_HARDWARE_CLOCK`)
For example, if we got timestamp 1234 from `depth_frame.get_timestamp()` and timestamp 1234 from `color_frame.get_timestamp()` do they correspond to the same moment or each sensor has its own clock?

Thank you.

MartyG-RealSense, 2021-03-22:
Hi @Egor18   A good starting point for answering your question would be the comment in these links:

https://github.com/IntelRealSense/librealsense/issues/2224#issuecomment-412504223

https://github.com/IntelRealSense/librealsense/issues/4551#issuecomment-532980040

The link below expands upon some of the principles the above link discusses.

https://github.com/IntelRealSense/librealsense/issues/1957#issuecomment-400715579

Egor18, 2021-03-23:
Thanks @MartyG-RealSense,

So I assume that there are 2 different clocks for depth and color.
But are there any guarantees on them being synched/launched at the same time?

Because here for example `_camera_time_base` is initialized no matter what type of frame it is:
https://github.com/IntelRealSense/realsense-ros/blob/4c52e767a1c4d0cf2fcd84bf6c577f6097c723bb/realsense2_camera/src/base_realsense_node.cpp#L1691

`_camera_time_base` is then used to calculate system time for every other frame, but if there are 2 completely independent clocks there should be 2 different base times.

MartyG-RealSense, 2021-03-23:
I answer cautiously as advanced metadata principles is outside of my knowledge.  So I will point you towards further answers from RealSense team members with greater specialist knowledge in this particular subject.

https://github.com/IntelRealSense/librealsense/issues/7016

https://github.com/IntelRealSense/librealsense/issues/1548

https://github.com/IntelRealSense/librealsense/issues/2188#issuecomment-409985803

Thanks for your patience in this case!