realsenseai/librealsense #8644: D435 distance example is wrong [recipe for target 'all' failed]

b2992872, 2021-03-23:
I tried to execute the distance example of D435 but the following error occurred, cmake was compiled successfully, and an error occurred during make

[image]
[image]

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | RealSense viewer 2.16.0)| 
| Operating System & Version |    Linux (Ubuntu 18)   | 
| Kernel Version (Linux Only)    | 4.14                                      | 
| Platform                                 | NVIDIA Jetson 
| SDK Version                            | 2.0                         | 
| Language                            |  C                        | 
| Segment			|       example                 | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I tried to execute the distance example of D435 but the following error occurred, cmake was compiled successfully, and an error occurred during make

MartyG-RealSense, 2021-03-23:
Hello, it looks as though you are building **rs-distance** as a C++ program (.cpp), but this example is a 'C' program (a different programming language).

b2992872, 2021-03-23:
set_target_properties(rs-distance PROPERTIES LINKER_LANGUAGE C)

MartyG-RealSense, 2021-03-23:
I will seek the advice of another RealSense team member on this particular subject.

@ev-mp  Do you have any advice that you could offer @b2992872  regarding problems with building the **rs-distance** C example using PROPERTIES LINKER_LANGUAGE C please?  Thanks!

b2992872, 2021-03-26:
This is the error I encountered in make

MartyG-RealSense, 2021-04-01:
@ev-mp Do you have any advice that you could offer @b2992872 regarding problems with building the rs-distance C example using PROPERTIES LINKER_LANGUAGE C please?  Thanks!

MartyG-RealSense, 2021-04-07:
Hi @b2992872  Do you still require assistance with the problem in this case please?  Thanks!

closed 2021-04-13 by MartyG-RealSense