realsenseai/librealsense #8646: D4 Vision Processor

nikhilarane, 2021-03-23:
Hello,

The D4 Vision Processor says that it can support up to 5 cameras. I have the following questions 

1) Can I get 2 depth streams (from 2 RealSense depth modules) connected to a single D4 vision processor?

2) Can I connect a custom stereo camera to the D4 vision processor?

3) Does Intel offer custom camera configurations for e.g. by using a more sensitive sensor for night time stereo applications?

Thanks,
Nikhil

MartyG-RealSense, 2021-03-23:
Hi @nikhilarane  I will refer these particular questions to Intel for their guidance.  Thanks for your patience!

RealSenseSupport, 2021-03-23:
Hi @nikhilarane 
1. No. The D4 VPU chip is able to connect to 5 image sensors, but the current ASIC card is designed to connect to one depth module only.
2. No.
3. Usually no, unless you place a big volume order. Please contact to RealSense sales team on customizing camera.

nikhilarane, 2021-03-25:
Thanks for your response @MartyG-RealSense.

closed 2021-03-25 by nikhilarane