realsenseai/librealsense #8647: D450 imagers IR sensitivity

nikhilarane, 2021-03-23:
Hello,

1) Is there anywhere I can find the spectral sensitivity of the sensors used in D450 module?

2) If I have an external IR illumination source. Will it be best to use 850nm or 950nm?

Thanks,
Nikhil

MartyG-RealSense, 2021-03-23:
1.  If you are asking about the IR frequencies that the infrared sensor of the 400 Series operates across, this information can be found on page 92 of the current edition of the data sheet document.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

A table from that page is quoted below:

[image]

If you are seeking the **spectral sensitivity curve** of the CMOS though then my recollection is that this particular information would be not available for public viewing.

Intel's white-paper document about **optical filters** has some useful curves about how filters affect the properties of light though.

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400

2.  The discussion in the link below provides technical advice about use of external IR illuminators with RealSense 400 Series cameras.

https://github.com/IntelRealSense/librealsense/issues/2000

850 nm is recommended by the conclusion section of Intel's white-paper document about projectors.

https://dev.intelrealsense.com/docs/projectors#section-11-conclusions

Quote: "It is possible to use light in visible or near-infrared (ranging from ~400-1000nm), although we normally recommend 850nm as being a good compromise in being invisible to the human eye while still being visible to the IR sensor with about 15-40% reduction in sensitivity, depending on the sensor used".

In a discussion about IR frequencies though it is advised that 940 nm can be used, but with reduced sensitivity.

https://github.com/IntelRealSense/librealsense/issues/1070#issuecomment-465610225

nikhilarane, 2021-03-25:
Thanks @MartyG-RealSense

closed 2021-03-25 by nikhilarane

MartyG-RealSense, 2021-03-25:
Thanks very much @nikhilarane for the updates!