realsenseai/librealsense #8649: How to get depth value from saved .png img?

fwf-lernen, 2021-03-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.11.00 | 
| Operating System & Version |   Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.15.0 | 
| Platform                                 | PC  |
| SDK Version                            |  2.42.0 | 
| Language                            | C | 

### Issue Description
Like https://github.com/IntelRealSense/librealsense/issues/5546#issue-542727255 ,I have saved depth image by 
`cv::Mat matDepth(cv::Size(imgWidth,imgHeight) , CV_16UC1 , (void *)depth.get_data() , cv::Mat::AUTO_STEP)`
However I have no idea how to get actual distance from saved imgs through cv::imread.
In rs_sensor.h,
`RS2_FORMAT_Z16             , /**< 16-bit linear depth values. **The depth is meters is equal to depth scale * pixel value**. */`
So, is that means I just need to use cv::imread and multiply a depth scale factor? If so, how can I get the scale factor?

MartyG-RealSense, 2021-03-23:
Hi @fwf-lernen   The C++ script in the link below, as part of a discussion on retrieving depth values from a PNG image, may be useful to you:

https://github.com/IntelRealSense/librealsense/issues/2200#issuecomment-411774418

 uses **cv::imread** to access a PNG file and generate a 3D point cloud from it.

https://github.com/IntelRealSense/librealsense/issues/2488

An alternative approach to the problem is described in this link:

https://stackoverflow.com/questions/51728827/using-a-depth-map/51759748#51759748

However, when RealSense depth data is saved as a PNG, most of the depth information is lost.  It is therefore better if possible to save the data as a **.raw** format image file that preserves the depth detail.  The link below provides scripting for retrieving depth information from a .raw file.

https://github.com/IntelRealSense/librealsense/issues/2231#issuecomment-416814331

Instead of obtaining the depth information from a PNG image, there is also the option of converting the cv::mat back into a RealSense **rs2::frame**, as described in the link below.

https://github.com/IntelRealSense/librealsense/issues/2634

fwf-lernen, 2021-03-24:
Hi @MartyG-RealSense Thank you for your reply. It's really helpful.
But I still have a question. Because I am using my own code to save data, how can I save the data as a .raw format image file using C++ script ?

fwf-lernen, 2021-03-24:
@MartyG-RealSense Thank you very much! That's really helpful.

closed 2021-03-24 by fwf-lernen

MartyG-RealSense, 2021-03-24:
You are very welcome @fwf-lernen - thanks for the update!