realsenseai/librealsense #8651: No 2d frames if real sense is used from vmware workstation

50c, 2021-03-23:
| | |
|---|---|
|**librealsense**|2.42.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D415|
|**Serial Number**|829512061264|
|**Firmware Version**|05.11.06.250|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|944523023396|
|**Firmware Update Id**|944523023396|

The cam works on host system but not within vm. Also other usb web cameras are working properly from vm as well.
This log I received from real sense viewer:

05:16:36 [Info]  - librealsense version: 2.42.0

05:16:36 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2588 - Framebuffer size changed to 1342 x 672
05:16:36 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2588 - Window size changed to 1342 x 672
05:16:36 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2588 - Scale Factor is now 1
05:16:37 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\context.cpp:381 - Found 1 RealSense devices (mask 254)
05:17:34 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\notifications.cpp:541 - Intel RealSense D415      (S/N 829512061264)
Current Version: 05.11.06.250
Recommended Version: 5.12.11.0
05:17:34 [Info]  - Intel RealSense D415 was selected as a default device
05:17:34 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\types.h:315 - get_video_proc()->GetRange(pu.property, &minVal, &maxVal, &steppingDelta, &defVal, &capsFlag) returned: HResult 0x80070490: "Element not found."
05:17:34 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\sensor.cpp:1112 - Failed to add Backlight Compensation control for RGB Camera
05:17:44 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\types.h:315 - get_camera_control()->GetRange(CameraControl_Exposure, &minVal, &maxVal, &steppingDelta, &defVal, &capsFlag) returned: HResult 0x80070490: "Element not found."
05:17:44 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\sensor.cpp:1112 - Failed to add Exposure control for RGB Camera
05:17:46 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\types.h:315 - get_video_proc()->GetRange(pu.property, &minVal, &maxVal, &steppingDelta, &defVal, &capsFlag) returned: HResult 0x80070490: "Element not found."
05:17:46 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\sensor.cpp:1112 - Failed to add Gamma control for RGB Camera
05:17:48 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\types.h:315 - get_video_proc()->GetRange(pu.property, &minVal, &maxVal, &steppingDelta, &defVal, &capsFlag) returned: HResult 0x80070490: "Element not found."
05:17:48 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\sensor.cpp:1112 - Failed to add Hue control for RGB Camera
05:17:50 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\types.h:315 - get_video_proc()->GetRange(pu.property, &minVal, &maxVal, &steppingDelta, &defVal, &capsFlag) returned: HResult 0x80070490: "Element not found."
05:17:50 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\sensor.cpp:1112 - Failed to add Saturation control for RGB Camera
05:17:52 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\types.h:315 - get_video_proc()->GetRange(pu.property, &minVal, &maxVal, &steppingDelta, &defVal, &capsFlag) returned: HResult 0x80070490: "Element not found."
05:17:52 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\sensor.cpp:1112 - Failed to add Sharpness control for RGB Camera
05:17:54 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\types.h:315 - get_video_proc()->GetRange(pu.property, &minVal, &maxVal, &steppingDelta, &defVal, &capsFlag) returned: HResult 0x80070490: "Element not found."
05:17:54 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\sensor.cpp:1112 - Failed to add White Balance control for RGB Camera
05:17:58 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\types.h:315 - hr returned: HResult 0x80070490: "Element not found."
05:17:58 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\sensor.cpp:1112 - Failed to add Enable Auto Exposure control for RGB Camera
05:18:02 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\types.h:315 - hr returned: HResult 0x80070490: "Element not found."
05:18:02 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\sensor.cpp:1112 - Failed to add Enable Auto White Balance control for RGB Camera
05:19:08 [Info]  - Starting streaming of Color...
05:19:08 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\sensor.cpp:1413 - Request: RGB8 Color, 
Resolved to: YUYV Color, 
07:11:59 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\output-model.cpp:869 - Unrecognized command ''

UI shows "no frame receiving"
Is it a way how we could deal with this issue?

Thank you

MartyG-RealSense, 2021-03-23:
Hi @50c   May I ask which version of VMWare Workstation you are using please - the free **Player** version or the **Pro** version?

I note that you are able to use USB webcams successfully with your VMWare, indicating that the USB emulation is functioning.

The information in the header of this case indicates that your D415 is using firmware driver version **5.11.6.250**.  Ideally, RealSense SDK version **2.42.0** is recommended to be matched with the recent firmware version **5.12.11.0**, though I also note that you have been able to successfully use the D415 on your host machine with your currently installed 5.11.6.250 firmware.

As a starting point in investigating your case, if you are using the Pro version then please check whether adding the D415 as a **Removable Device** resolves the errors.

https://docs.vmware.com/en/VMware-Workstation-Pro/15.0/com.vmware.ws.using.doc/GUID-07883F0A-E7B5-4CBA-837D-0BA986E8B900.html

MartyG-RealSense, 2021-03-31:
Hi @50c  Do you require further assistance with this case, please?  Thanks!

50c, 2021-04-01:
Hi @MartyG-RealSense,  I'm sorry but I'm not succeeded with this problem. I have applied the instaractions you sent, but only the change was that some input devices like keyboard are then connectable to the vm system.
As you suggested, I've updated the RealSense firmware to the recent version but it didn't solve the problem.
I have updated the whole vm using the last version of VMWare format, but without luck.
I'm using VMWare Workstation Pro Version 15.5.5 build-16285975

Here is the output from RealSense Viewer:

00:43:54 [Info]  - librealsense version: 2.42.0

00:43:54 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2588 - Framebuffer size changed to 1342 x 649
00:43:54 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2588 - Scale Factor is now 1
00:43:55 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\context.cpp:381 - Found 1 RealSense devices (mask 254)
00:44:15 [None] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\notifications.cpp:541 - Frame Metadata is a device feature allowing
software synchronization between different
camera streams.
It must be explicitly enabled on Windows OS

00:45:01 [Info]  - Intel RealSense D415 was selected as a default device
00:45:02 [Info] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\src\types.h:1103 - USB CAM overflow
00:45:02 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\notifications.cpp:514 - USB CAM overflow
00:45:04 [Info] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\src\types.h:1103 - USB CAM overflow
00:45:06 [Info] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\src\types.h:1103 - USB CAM overflow
00:45:08 [Info] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\src\types.h:1103 - USB CAM overflow
00:45:10 [Info] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\src\types.h:1103 - Error polling loop is not behaving as expected!
This can indicate an issue with camera firmware or the underlying OS...
00:45:10 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\notifications.cpp:514 - Error polling loop is not behaving as expected!
This can indicate an issue with camera firmware or the underlying OS...
00:46:57 [Info]  - Starting streaming of Color...
00:46:57 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\sensor.cpp:1413 - Request: RGB8 Color, 
Resolved to: YUYV Color, 

[image]

Do you have any idea?

MartyG-RealSense, 2021-04-01:
Given that you are experiencing **USB CAM overflow** errors, suggesting that the USB system cannot cope with the demand of the incoming frames, could you try a lesser FPS speed or resolution please to see whether reducing the demand on the system enables the RGB frames to appear.  For example, 1280x720 at 15 FPS, or 640x480 at 30 FPS

50c, 2021-04-01:
Thank you, @MartyG-RealSense, I've tried different resolution settings. The highest I could use was 640x480 * 30. All higher resolutions even with lowest frame rate lead to the behavior described above. Is is the limitation of virtualization stack I have to stick with or there are some settings I could play with around? For the use case the desirable resolution would be 1280x720.
Thank you so far for your great support.

MartyG-RealSense, 2021-04-01:
You are very welcome.  :)

As the highest you could achieve was 640x480 at 30 FPS, it may be worth testing 1280x720 at 6 FPS.  As 1280x720 only supports 6 FPS on a **USB 2** connection, if it works then it might suggest that the camera is behaving like it is on a USB 2 connection instead of USB 3.2 when on VM.

[image]

50c, 2021-04-06:
Hi, using 1280x720 at 6 FPS there are still no frames. Most likely the issue isn't RealSense related and caused by some virtualization limitation I'm not yet able to control, so I give it up. Thank you for your support. I will close the issue

closed 2021-04-06 by 50c

MartyG-RealSense, 2021-04-06:
Thanks very much @50c for the update - good luck with your VM work!